MaJerle/lwesp #5: esp_set_server callback doesn't work

rafalo235, 2018-07-24:
Hello,
Apparently callback after creating server under 80 port doesn't work. I have configured separate esp_input_process thread with DMA buffer. Callback after esp_sta_join works fine, but once I call esp_set_server nothing happen. I checked system under gdb, and everything looks fine, data are present in dma buffer, but it is not processed. It worked under previous version of library - which were marked deprecated and removed, so I don't think it's AT firmware issue.
My code is pushed on github: https://github.com/rafalo235/nixie-clock-sw/tree/esp-lib
https://github.com/rafalo235/nixie-clock-sw/blob/esp-lib/src/connection-task/input-task.c - input task
https://github.com/rafalo235/nixie-clock-sw/blob/esp-lib/src/connection-task/connection-task.c - worker thread - line 72 is called, but Server_Callback is not called even if data are received
https://github.com/rafalo235/nixie-clock-sw/blob/esp-lib/inc/port/esp8266-at-lib/esp_config.h - esp config
esp_sys.h is coppied from cmsis sample

MaJerle, 2018-07-24:
Hello @rafalo235,

which callback exactly is not working anymore?

rafalo235, 2018-07-24:
@MaJerle 
Server_Callback registered with this:
      if ((res = esp_set_server(1, 80, ESP_CFG_MAX_CONNS, 100, &Server_Callback, 1)) == espOK) {
(line 72 connection-task.c)

MaJerle, 2018-07-24:
Correct. But which callback is not called anymore. Main one, `Connection` or `Server` callback?

rafalo235, 2018-07-24:
Server callback. Connection callback works fine

MaJerle, 2018-07-24:
I was not able to reproduce your error. Using example from link below, please find my application log.

Example project: https://github.com/MaJerle/ESP_AT_Lib/blob/master/examples/win32/server_rtos/main.c

Check carefully log below, you will see that callback for server is called as messages are printed to window.

```
Starting ESP application!
AT+RST

OK

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

î┬▄ΣéN∞|sé√N|∞îlÅps$c▓▄≡╛ä2s▓▄d▄NαéNΓ ll îΓrôlî≥2ä∞l`îΓsôlΣl2ä$`îΓsôlÅ≤îld`rdÄÉnî╛Bc∞|îBlB▄Γ£"pÇî£ÄéÄÉdcΣ√N∩ÇNo┬é▄ äd≥ΓNî£ädlîÅΣlÅÄ∞Σ∞ÄrdÇoⁿ î╛"cl$∞îÅ2cî╛"clîcrlrl
ready
Device reset detected!
ATE1

OK
AT+GMR
AT version:1.7.0.0-dev(Jun 19 2018 20:38:48)
SDK version:3.0.0-dev(c0f7b44)
compile time:Jun 28 2018 01:08:51
OK
AT+CWMODE=3

OK
AT+SYSMSG_CUR=3

OK
AT+CIPMUX=1

OK
AT+CWLAPOPT=1,2047

OK
AT+CIPSTATUS
STATUS:5

OK
AT+CIPAP_CUR?
+CIPAP_CUR:ip:"192.168.4.1"
+CIPAP_CUR:gateway:"192.168.4.1"
+CIPAP_CUR:netmask:"255.255.255.0"

OK
AT+CIPAPMAC_CUR?
+CIPAPMAC_CUR:"de:4f:22:18:22:56"

OK
AT+CIPDINFO=1

OK
Library initialized!

Scanning access points...
AT+CWLAP
+CWLAP:(3,"my_hidden_network",-39,"aa:6b:6e:58:cf:66",11,28,0,4,4,7,0)

OK
AP found: my_hidden_network
Connecting to "my_hidden_network" network...
AT+CWJAP_DEF="my_hidden_network","its private"
WIFI CONNECTED
WIFI GOT IP

OK
AT+CIPSTA_CUR?
+CIPSTA_CUR:ip:"172.20.10.2"
+CIPSTA_CUR:gateway:"172.20.10.1"
+CIPSTA_CUR:netmask:"255.255.255.240"

OK
AT+CIPSTAMAC_CUR?
+CIPSTAMAC_CUR:"dc:4f:22:18:22:56"

OK
Connected to my_hidden_network network!
Station IP address: 172.20.10.2
AT+CIPSERVERMAXCONN=5

OK
AT+CIPSERVER=1,80

OK
AT+CIPSTO=0

OK
+LINK_CONN:0,0,"TCP",1,"172.20.10.1",57894,80

+IPD,0,377,172.20.10.1,57894:GET / HTTP/1.1
HosConnection 0 active as server!
t: 172.20.10.2
Upgrade-Insecure-Requests: 1
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Mozilla/5.0 (iPhone; CPU iPhone OS 11_0_3 like Mac OS X) AppleWebKit/604.1.34 (KHTML, like Gecko) CriOS/67.0.3396.87 Mobile/15A432 Safari/604.1
Accept-Language: en-us
Accept-Encoding: gzip, deflate
Connection: keep-alive

Server connection 0 data received with 377 bytes
AT+CIPCLOSE=0
0,CLOSED

OK
Server connection 0 closed!
+LINK_CONN:0,0,"TCP",1,"172.20.10.1",57895,80

+IPD,0,377,172.20.10.1,57895:GET / HTTP/1.1
Host: 172.20.10.2
Upgrade-Insecure-Requests: 1
Accept: text/htConnection 0 active as server!
ml,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Mozilla/5.0 (iPhone; CPU iPhone OS 11_0_3 like Mac OS X) AppleWebKit/604.1.34 (KHTML, like Gecko) CriOS/67.0.3396.87 Mobile/15A432 Safari/604.1
Accept-Language: en-us
Accept-Encoding: gzip, deflate
Connection: keep-alive

Server connection 0 data received with 377 bytes
AT+CIPCLOSE=0
0,CLOSED

OK
Server connection 0 closed!
+LINK_CONN:0,0,"TCP",1,"172.20.10.1",57896,80

+IPD,0,377,172.20.10.1,57896:GET / HTTP/1.1
Host: 172.20.10.2
Upgrade-Insecure-RequConnection 0 active as server!
ests: 1
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Mozilla/5.0 (iPhone; CPU iPhone OS 11_0_3 like Mac OS X) AppleWebKit/604.1.34 (KHTML, like Gecko) CriOS/67.0.3396.87 Mobile/15A432 Safari/604.1
Accept-Language: en-us
Accept-Encoding: gzip, deflate
Connection: keep-alive

Server connection 0 data received with 377 bytes
AT+CIPCLOSE=0
0,CLOSED

OK
Server connection 0 closed!
```

rafalo235, 2018-07-24:
I see some mismatch in AT responses, I think, that `AT+CIPSERVERMAXCONN=5` command is not recognized correctly. I will try to update AT firmware 
```
Ai-Thinker Technology Co.,Ltd.

ready
ATE0\r

OK
AT version:0.40.0.0(Aug  8 2015 14:45:58)
SDK version:1.3.0
Ai-Thinker Technology Co.,Ltd.
Build:1.3.0.2 Sep 11 2015 11:48:04
OK

OK

ERROR

ERROR

OK

ERROR
STATUS:5

OK
+CIPAP_CUR:ip:\"192.168.4.1\"
+CIPAP_CUR:gateway:\"192.168.4.1\"
+CIPAP_CUR:netmask:\"255.255.255.0\"

OK
+CIPAPMAC_CUR:\"62:01:94:06:56:00\"

OK

OK
WIFI CONNECTED
WIFI GOT IP

OK
+CIPSTA_CUR:ip:\"192.168.128.68\"
+CIPSTA_CUR:gateway:\"192.168.128.1\"
+CIPSTA_CUR:netmask:\"255.255.255.0\"

OK
+CIPSTAMAC_CUR:\"60:01:94:06:56:00\"

OK

ERROR

OK

OK
0,CONNECT

+IPD,0,438,192.168.128.64,52120:GET /index.html HTTP/1.1
Host: 192.168.128.68
Connection: keep-alive
Cache-Control: max-age=0
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/67.0.3396.99 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8
Accept-Encoding: gzip, deflate
Accept-Language: pl-PL,pl;q=0.9,en-US;q=0.8,en;q=0.7
```

MaJerle, 2018-07-24:
Yes, version is not official from Espressif.

http://majerle.eu/documentation/esp_at/html/page_faq.html

closed 2018-07-24 by MaJerle

rafalo235, 2018-07-24:
@MaJerle Checked with software 1.6.2, and it works :) thanks for help. However I'm curious why old at library worked with that one.