MaJerle/lwesp #6: Improvement for "ports"

sn00pster, 2018-07-30:
I've just discovered your library and am getting it working under FreeRTOS (plain freertos, not using the cmsis layer). Something that I've noticed is that you've decided to use defines to select the "OS" port. 

Was there any reason for this? In my mind it would be much cleaner to just have a folder for each port and keep the file names for each port the same, you then just include the correct files to your project and set the correct include path for the platform.

Reason for me asking is that to add a platform you currently have to modify the source of the library, if you do it the other way around then you don't, you just add the correct files to your project and you can use it and update the library without causing yourself any issues.

If you're not keen on that, then maybe a third OS option could be "USER", where you can select that in the config and it uses files for the port suffixed with user, that way you can keep it as you currently do and people can do their own ports and can update without having to make changes to the underlying library.

MaJerle, 2018-07-30:
This is a good proposal and I implemented it just now. It adds new macro you may set for configuration.

https://github.com/MaJerle/ESP_AT_Lib/commit/3486c572c189dca4c72284140f0529b04a747520

Updated documentation is also available:
http://majerle.eu/documentation/esp_at/html/group___e_s_p___s_y_s.html#ESP_SYS_PORTS

sn00pster, 2018-07-30:
Awesome!  Will make life easier for those maintaining "unofficial" ports of the library.

Thanks for taking the suggestion and applying it so quickly!

MaJerle, 2018-07-31:
If you have operating system port (native freertos) or ChibiOs or similar, feel free to create pull request.

closed 2018-07-31 by MaJerle