MaJerle/lwesp #9: False positive return code

sn00pster, 2018-08-03:
I need to investigate this, but I was playing with baud rates earlier (seeing what I could go up to) and the following code:

```
esp_init(esp_callback_func, 1);
    
esp_set_at_baudrate(921600*2, 1);

//set_baud(921600*2);`

if (esp_sta_join("SSID", "PASSWORD", 0, 0, 1)==espOK)
    print("joined");
else
    print("failed");
```

esp_sta_join was returning espOK when the baud rate of the ESP was changed but not the baud rate of the board, I confirmed by debug that the baud rate change on the ESP was taking place and that after that no more comms were taking place.

I would expect the esp_sta_join to return a failure code.

MaJerle, 2018-08-04:
Thanks. If you find where is problem, let me know, or please do pull request.

sn00pster, 2018-08-05:
My fault, I've misread the return code from my rtos port, I will submit a pull request when I'm a bit more awake!

But, it has lead me to something else:

```
if (block && res == espOK) {                /* In case we have blocking request */
        uint32_t time;
        **time = esp_sys_sem_wait(&msg->sem, max_block_time * 100);   /* Wait forever for semaphore access for max block time */**
        if (ESP_SYS_TIMEOUT == time) {          /* If semaphore was not accessed in given time */
            res = espTIMEOUT;                   /* Semaphore not released in time */
        } else {
            res = msg->res;                     /* Set response status from message response */
        }
        if (esp_sys_sem_isvalid(&msg->sem)) {   /* In case we have valid semaphore */
            esp_sys_sem_delete(&msg->sem);      /* Delete semaphore object */
            esp_sys_sem_invalid(&msg->sem);     /* Invalidate semaphore object */
        }
        ESP_MSG_VAR_FREE(msg);                  /* Release message */
    }
```

The return of this is whether or not the "time" elapsed and not whether or not the semaphore was taken, this leads to the possibility of a situation where an unfortunate lengthy delay but successful taking of the semaphore could signal a fail when infect the semaphore was taken.

MaJerle, 2018-08-05:
Can you elaborate more? Waiting for signal unlimited time if esp reply fails may lead to forever blocked thread. Am I missing something?

sn00pster, 2018-08-05:
Sorry, I may be getting confused, I'm very tired.  Ignore me on that one, sorry, I misread the code.

I will submit my pull request tomorrow morning when I have push my changes.

closed 2018-08-08 by MaJerle