MaJerle/lwesp #10: Documentation on Firmware

sn00pster, 2018-08-05:
Hi,

The documentation could do with a bit more of a hint/warning about the firmware that people have on their device, it absolutely requires the official firmware from ESP to work properly (I couldn't get the http server working, so spent ages figuring out how to flash the ESP firmware on it)

After flashing, the HTTP server works correctly.    While bits may work fine with the AI-THINKER firmware not all of it will as I found out!

MaJerle, 2018-08-05:
http://majerle.eu/documentation/esp_at/html/page_faq.html

sn00pster, 2018-08-05:
I think it could do with being repeated here:

http://majerle.eu/documentation/esp_at/html/page_update_at_software.html#sect_update_process_preparations

and more importantly in the readme.md

I missed the FAQ...I suspect others might too.

MaJerle, 2018-08-06:
Now I added new event, which is triggered if ESP AT software version is less than minimum supported by library. Check examples, main.c files.

sn00pster, 2018-08-08:
Cool, always best to make it as explicit as possible what's required!

closed 2018-08-08 by sn00pster