MaJerle/lwesp #12: HTML wrong in example data?

sn00pster, 2018-08-08:
Hi,

I don't know enough about the frameworks being used here, but the demo suggests that there should be a message written to a div dynamically via javascript after the page is loaded.

However, the div with the id "maindiv" doesn't exist, therefore the message never appears.

I fixed this with:

```
static const uint8_t
responseData[] = ""
    "HTTP/1.1 200 OK\r\n"
    "Server: " HTTP_SERVER_NAME "\r\n"
    "Content-Type: text/html\r\n"
    "\r\n"
    "<html>\n"
    "   <head>\n"
    "       <title><!--#title--></title>\n"
    "       <meta http-equiv=\"Refresh\" content=\"1\" />\n"
    "       <script src=\"https://ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js\"></script>\n"
    "       <script src=\"/js/js.js\" type=\"text/javascript\"></script>\n"
    "       <link rel=\"stylesheet\" href=\"https://maxcdn.bootstrapcdn.com/bootstrap/4.0.0-beta.2/css/bootstrap.min.css\" integrity=\"sha384-PsH8R72JQ3SOdhVi3uxftmaW6Vc51MKb0q5P2rRUpPvrszuE4W1povHYgTpBfshb\" crossorigin=\"anonymous\" />\n"
    "       <link rel=\"stylesheet\" type=\"text/css\" href=\"/css/style.css\" />\n"
    "   </head>\n"
    "   <body>\n"
    "       <div class=\"container\">\n"
    "           <h1>Welcome to web server hosted by ESP8266 Wi-Fi module!</h1>\n"
    "       </div>\n"
    "       <div id=\"maindiv\"/>"
    "       <footer>\n"
    "           <div class=\"container\">\n"
    "               Copyright &copy; 2017. All rights reserved. Webserver is hosted on ESP8266.\n"
    "           </div>\n"
    "       </footer>\n"
    "   </body>\n"
    "</html>\n";

```

But as I said, I don't know enough about the frameworks being used here as to whether they themselves create divs on the fly, but with this modification it does work how I would expect it to.

MaJerle, 2018-08-08:
To which example are you referring to? I don't see an example to dynamically load any data. (but I might be wrong!).

sn00pster, 2018-08-08:
ESP_AT_Lib/src/apps/http_server/esp_http_server_fs.c

This data file, dynamically writes to the page after load (not dynamically loads data), except the div it writes to is missing from the main page html.

MaJerle, 2018-08-08:
Ok. Suppose of this file is not to have it final app, but to `fake` file system.

closed 2018-08-08 by MaJerle

sn00pster, 2018-08-08:
Absolutely, but it's a starting point of code and it's wrong.