MaJerle/lwesp #24: esp_sys_mbox_putnow implemention in freertos port

xiongyu0523, 2018-12-20:
In freertos system porting file, why xQueueSendFromISR is called? I think all APIs are in task context.. 
From the document I can see the only difference between put and putnow is whether timeout is used. 

uint8_t
esp_sys_mbox_putnow(esp_sys_mbox_t* b, void* m) {
    freertos_mbox mb;

    mb.d = m;
    return xQueueSendFromISR(*b, &mb, 0) == pdPASS;
}

MaJerle, 2018-12-20:
FreeRTOS port was developed as pull request. Putnow must be capable to work under interrupt aswell.

xiongyu0523, 2018-12-21:
Have you tested this pull request?
If it is intended to be used under interrupt context, then we need:

1. check the context to decide whether xQueueSend or xQueueSendFromISR will be used
2. If xQueueSendFromISR is used, then we need to check the 3rd parameters to notify scheduler.

xiongyu0523, 2018-12-21:
BTW, I think esp_sys_mbox_put() is not expected to be called in interrupt context, right?

MaJerle, 2018-12-21:
@xiongyu0523 I haven't tested it. Library itself never uses any API under interrupt context. Lib doesn't care about platform, you may run it on win/linux/MCU/etc. If interrupts are used depends on implementation. If CPU has DMA + separate thread for data receive, there is no need for interrupt context.

I never use OS native functions, I'm only using ARM in embedded apps and I always use CMSIS-OS layer on top. There I can use single function `osMessagePut(ptr, val, timeout)` and function will check either it is IRQ or non-IRQ context.

I let you update this port.

xiongyu0523, 2018-12-21:
Hi, Majerle

Thanks for your information. I do not have STM32 board. I'm currently porting the lib to i.MX RT silicon platform and uses FreeRTOS native API. I can update the freertos port and raise a pull merge to integrate the i.mx ll port once my setup works.

MaJerle, 2018-12-21:
I suggest you create new port and name it esp_sys_freertos_arm_cortex_m and use all functions from ARM CMSIS to determine if you are in irq context or not.

Would it work for you?

MaJerle, 2018-12-24:
You are right, general FreeRTOS port won't work for any arch. Only function which (by library) may be called from interrupt is `esp_sys_mbox_putnow`, when using `esp_input`. If `esp_input` is not used, there is no need for interrupt capable `mbox_putnow` function.

xiongyu0523, 2018-12-26:
OK. I understand now. 

My code is almost ready, the question is how to deal with arch, can we put a macro in esp_sys_freertos_os.h and let the user modify if they're not on ARM? More than one FreeRTOS port looks weird. 

```
#include "core_cm7.h"
#define __isIRQContext()            __get_IPSR()
```

closed 2018-12-31 by MaJerle