MaJerle/lwesp #25: Cyclic attempts to restart

Nuck-TH, 2018-12-24:
I'm using your library on stm32l429-nucleo board.
If i follow same LL-init sequence like in stm32 examples(DMA is started before pulling RST up) - startup info end up in the buffer before return from LL-init function and it causes stack to send restsrt cmd over and over again, because it apparently thinks that restart wasn't forced.

For now i circumvent it by waiting long enough to let module spit startup info before starting DMA, but i think that this case(startup info after LL init function) should be handled in stack.

MaJerle, 2018-12-24:
Stack doesn't care about dummy data from module, it waits for `ready\r\n` state. I'm not sure how can I reproduce your effect on my hardware.

Nuck-TH, 2018-12-24:
If i follow code logic correct - this case handled like random reset in the middle of operation, while it just startup. espi_reset_everything sends reset AT command, and for some reason configuration commands are proceed to be sended without wait for ready again which messes up input and this goes on until some delay occurs. DMA buffer is quite large(4096 bytes), so even if RTS doesn't work(which i will check) - there should not be overflow.
Maybe "ready" is mangled in attempt of unciode decode? because startup info is garbage(thanks to 26Mhz crystal in ESP-12, so startup info is sent at 74880baud).

>     1. Enable DMA RX
>     2. Pull RST low, wait 10ms
>     3. Pull RST high, wait 200ms
>     4. Exit from function

This is how i did so it restarts.
It works normal when as follows:

1. Pull RST low, wait 10ms
2. Pull RST high, wait 1000ms
3. Enable DMA RX

MaJerle, 2018-12-24:
After checking the code, it looks like it should work ok, because, even if first time `ready\r\n` was detected as not forced reset (which was in fact not forces by stack, but by user), function `espi_reset_everything` will get parameter `forced = 0` and will manually reset it in non-blocking mode. From here, any `ready` event is detected as forced because we are in reset mode.

Do you have latest pull from the lib?

Nuck-TH, 2018-12-24:
last release

MaJerle, 2018-12-24:
> last release

Pull last commit.

MaJerle, 2018-12-24:
This problem is weird. Do you use `esp_input` or `esp_input_process` function to process input data? If first, then it may be an issue, if second, then there is not issue. In `esp_input` I forgot to add check if is initialized, while in `esp_input_process` this is included. This means that if your `ll_init` function delays and ESP sends data and you process them in separate thread and try to send it to stack, stack will ignore  your received data, meaning `ready` from GPIO reset is ignored.

MaJerle, 2018-12-25:
Logic is ok. when you call `esp_reset(0)`, command is added to queue in non-blocking mode. When it will be executed, then it will wait for `ready` message. No matter if I enable or disable RESET_ON_INIT, I'm not able to reproduce your behavior, no matter what I do.

closed 2019-08-02 by MaJerle