MaJerle/lwesp #26: Semaphore（sem_sync ） multiple meaningless calls

zhangxichao, 2019-01-03:
Hi MaJerle:

In Esp.c File:

    /* Create threads */
    esp_sys_sem_wait(&esp.sem_sync, 0);         /* ①*/
    if (!esp_sys_thread_create(&esp.thread_producer, "esp_producer", esp_thread_producer, &esp.sem_sync, ESP_SYS_THREAD_SS, ESP_SYS_THREAD_PRIO)) {
        esp_sys_sem_release(&esp.sem_sync);     /* Release semaphore */
        goto cleanup;
    }
    esp_sys_sem_wait(&esp.sem_sync, 0);         /* ② */
    if (!esp_sys_thread_create(&esp.thread_process, "esp_producer", esp_thread_process, &esp.sem_sync, ESP_SYS_THREAD_SS, ESP_SYS_THREAD_PRIO)) {
        esp_sys_sem_release(&esp.sem_sync);     /* Release semaphore */
        goto cleanup;
    }
    esp_sys_sem_wait(&esp.sem_sync, 0);         /* ③ */
    esp_sys_sem_release(&esp.sem_sync);         /* Release semaphore */

①：A sem_sync lock is obtained,If the producer thread is created successfully 
②：It will try to get the lock again,If the process thread is created successfully，But no one released the lock，So will this lock fail?
③：try to get the lock again，Fail?

And in the producer thread, why continue to acquire this lock?

I'm puzzled. Please give me some advice. Thank you.

zhangxichao, 2019-01-03:
/* Create threads */
esp_sys_sem_wait(&esp.sem_sync, 0);         /* ①*/
if (!esp_sys_thread_create(&esp.thread_producer, "esp_producer", esp_thread_producer, &esp.sem_sync, ESP_SYS_THREAD_SS, ESP_SYS_THREAD_PRIO)) {
    esp_sys_sem_release(&esp.sem_sync);     /* Release semaphore */
    goto cleanup;
}
esp_sys_sem_wait(&esp.sem_sync, 0);         /* ② */
if (!esp_sys_thread_create(&esp.thread_process, "esp_producer", esp_thread_process, &esp.sem_sync, ESP_SYS_THREAD_SS, ESP_SYS_THREAD_PRIO)) {
    esp_sys_sem_release(&esp.sem_sync);     /* Release semaphore */
    goto cleanup;
}
esp_sys_sem_wait(&esp.sem_sync, 0);         /* ③ */
esp_sys_sem_release(&esp.sem_sync);         /* Release semaphore */

①：A sem_sync lock is obtained,If the producer thread is created successfully
②：It will try to get the lock again,If the process thread is created successfully，But no one released the lock，So will this lock fail?
③：try to get the lock again，Fail?

And in Line 89 to 91 of Esp_threads.c File，
the producer thread, why continue to acquire this lock?

MaJerle, 2019-01-03:
I don't see your line numbers, anyway, last wait and release is not necessary, but it is also not a fail. Code works as expected, no bugs here.

closed 2019-01-03 by MaJerle

zhangxichao, 2019-01-03:
Well, I just didn't notice that the thread entry unlocked the semaphore.
But what's the meaning of Marker ③ and why?
And in Line 89 to 91 of Esp_threads.c File，
the producer thread, why continue to acquire this lock?

the  wait and Release of Marker ③ and the wait of Line 89 to 91 of Esp_threads.c, 
Can these three statement be removed?

MaJerle, 2019-01-03:
I don't know what lines 89-91 are, sorry. In my case these are semaphore create, by last commit. Can you please clarify what are you interested?

if you are interested in:

    esp_sys_sem_wait()
    esp_sys_sem_release();

Then answer is no. It cannot be removed as we have to wait first to thread to unlock semaphore. Once this happens, we will pass wait function but then since noone else will release semaphore, we have to do it manually.

zhangxichao, 2019-01-03:
Line 89 ~ 91 is:

            ESP_CORE_UNPROTECT();
            esp_sys_sem_wait(&e->sem_sync, 0);
            ESP_CORE_PROTECT();

zhangxichao, 2019-01-03:
Why wait for threads to release semaphores in order to ensure that threads are running?

MaJerle, 2019-01-03:
@zhangxichao, Why would be fail to acquire lock again? Semaphores are used to sync threads. One thread is waiting another to finish. This is happening in esp_threads.c.

zhangxichao, 2019-01-04:
Hi,MaJerle
Why not put 
        ESP_CORE_UNPROTECT();
        esp_sys_sem_wait(&e->sem_sync, 0);
        ESP_CORE_PROTECT();
after Mark ③?

MaJerle, 2019-01-04:
For further discussion, open new issue and download latest changes first.