MaJerle/lwesp #27: sem-sync

zhangxichao, 2019-01-04:
Hi MaJerle,

/* Create threads */
    esp_sys_sem_wait(&esp.sem_sync, 0);         /* Lock semaphore */
    if (!esp_sys_thread_create(&esp.thread_produce, "esp_produce", esp_thread_producer, &esp.sem_sync, ESP_SYS_THREAD_SS, ESP_SYS_THREAD_PRIO)) {
        esp_sys_sem_release(&esp.sem_sync);     /* Release semaphore */
        goto cleanup;
    }
    esp_sys_sem_wait(&esp.sem_sync, 0);         /* Wait semaphore, should be unlocked in producer thread */
    if (!esp_sys_thread_create(&esp.thread_process, "esp_process", esp_thread_process, &esp.sem_sync, ESP_SYS_THREAD_SS, ESP_SYS_THREAD_PRIO)) {
        esp_sys_sem_release(&esp.sem_sync);     /* Release semaphore */
        goto cleanup;
    }
    esp_sys_sem_wait(&esp.sem_sync, 0);         /* Wait semaphore, should be unlocked in producer thread */
    esp_sys_sem_release(&esp.sem_sync);         /* Release semaphore */

zhangxichao, 2019-01-04:
Why not put
          ESP_CORE_UNPROTECT();
          esp_sys_sem_wait(&e->sem_sync, 0);
          ESP_CORE_PROTECT();         /*File esp_thread.c, Line 89-91*/
after last esp_sys_sem_release?

Because what we need now is the synchronization of semaphores.

MaJerle, 2019-01-04:
Why do you think is needed?

MaJerle, 2019-01-04:
Ok, so what do we gain with your proposal over mine imolementation? I do not understand your point well.

zhangxichao, 2019-01-04:
ESP_CORE_UNPROTECT();               /* Release protection, think if this is necessary, probably shouldn't be here */
            esp_sys_sem_wait(&e->sem_sync, 120000); /* Lock semaphore, should be unlocked before! */
            ESP_CORE_PROTECT();                 /* Protect system again, think if this is necessary, probably shouldn't be here */
            res = msg->fn(msg);                 /* Process this message, check if command started at least */
            if (res == espOK) {                 /* We have valid data and data were sent */
                ESP_CORE_UNPROTECT();           /* Release protection */
                time = esp_sys_sem_wait(&e->sem_sync, msg->block_time); /* Wait for synchronization semaphore */
                ESP_CORE_PROTECT();              /* Protect system again */
                esp_sys_sem_release(&e->sem_sync);  /* Release protection and start over later */
                if (time == ESP_SYS_TIMEOUT) {  /* Sync timeout occurred? */
                    res = espTIMEOUT;           /* Timeout on command */
                }
            } else {
                esp_sys_sem_release(&e->sem_sync);  /* We failed, release semaphore automatically */
            }

So what does esp_sys_sem_wait (&e->sem_sync, 120000) mean?
To make the semaphore zero, prepare for synchronization with the process thread.
I don't understand right?

MaJerle, 2019-01-04:
I will give you now full answer, but first please carefully read following points:

1. Check how I updated your comments here, to make your code in comment visible normally. More info on Github help: https://help.github.com/articles/creating-and-highlighting-code-blocks/
2. Since you did not download latest GIT changes (`git pull`) as proposed, I will manually refer now to latest commit file. For historical purposes, it is located on URL below. **All line numbers are from this file directly on the link**. https://github.com/MaJerle/ESP_AT_Lib/blob/e6f1c32df67923e43b949d880bb3663d60bcc856/src/esp/esp_threads.c
3.  Use documentation for your reference point on inter-thread communication. https://majerle.eu/documentation/esp_at/html/page_appnote.html#sect_thread_comm
4. Semaphores in AT-Lib are binary only, always used for thread synchronization purposes, nothing else.

Back to topic.
1. Lines 90 waits for semaphore. This operation must be executed instantly, otherwise there is serious error in the system for some reason. Here, I could add check if there is any kind of timeout, report serious error and do not allow any other command to proceed.
2. After semaphore is locked, we start sending first command to AT port.
3. After first command has been sent, we try to lock semaphore again (line `95`).. Remember, we cannot lock it again as it was already locked by us, unless someone will release it. Release happens in another function, which, in order to work properly, needs mutual exclusion, thus we have to call `esp_core_unlock()` before we try to lock semaphore (line `94`).
4. Release happens on line `930` here: https://github.com/MaJerle/ESP_AT_Lib/blob/e6f1c32df67923e43b949d880bb3663d60bcc856/src/esp/esp_int.c
5. If lock was successful (command finished), we have to manually release semaphore back to default state (line `100`) and start over for next command.

closed 2019-01-04 by MaJerle

zhangxichao, 2019-01-05:
```
12345
```