MaJerle/lwesp #32: esp_netconn_close() crashes

ev-ermakov, 2019-02-02:
Hello,

In `esp_netconn_close()`  (esp_netconn.c on lines 697-698):

```c
    esp_conn_set_arg(conn, NULL);               /* Reset argument */
    esp_conn_close(conn, 1);                    /* Close the connection */
```

Meanwhile, in `netconn_evt()` on lines 178-185:

```c
        case ESP_EVT_CONN_RECV: {
            esp_pbuf_p pbuf;

    (1)     nc = esp_conn_get_arg(conn);        /* Get API from connection */
            pbuf = esp_evt_conn_recv_get_buff(evt); /* Get received buff */

            esp_conn_recved(conn, pbuf);        /* Notify stack about received data */
    (2)     nc->rcv_packets++;                  /* Increase number of received packets */
            ...
```

Suppose there is a connection to a host that is actively transmitting data to us (i. e. with a high bit rate). If the `esp_netconn_close()` function is called at this moment, the program will crash.

Screenshots: exception and stack trace.

[image]
[image]

MaJerle, 2019-02-02:
Thanks for reporting. Commit 8368226c3fc54c902c23dd6082c56397ebbb0068 should fix the issue.

ev-ermakov, 2019-02-02:
Thank you. The program no longer crashes.

closed 2019-02-02 by MaJerle