MaJerle/lwesp #33: Return value of esp_netconn_connect()

ev-ermakov, 2019-02-02:
The `esp_netconn_connect()` function returns `espOK`, even if there is no connection to the Wi-Fi access point.

Is this a bug or I just use it incorrectly?

By the way, it returns `espOK`, even if there is a WiFi link, but the remote host does not exist (or when the port is closed).

--
Sorry for my English

MaJerle, 2019-02-02:
1. You are right, this has not been fully implemented yet, now added with commit https://github.com/MaJerle/ESP_AT_Lib/commit/edba5a53b58aba61808224cb1172321b817d6624
2. You are wrong, `ERROR` event is triggered when there is error connecting to server. Makes sense to use it if you start connection in non-blocking mode. Netconn API is blocking.

ev-ermakov, 2019-02-02:
2. I'm not sure I understood you.

Try to run:
```c
    esp_sta_join( "<ssid>", "<pass>", NULL, 0, NULL, NULL, 1 );

    esp_netconn_p ncs[10];
    for (size_t i = 0; i < ESP_ARRAYSIZE(ncs); i++) {
        ncs[i] = esp_netconn_new(ESP_NETCONN_TYPE_TCP);
        if (ncs[i] != NULL) {
            espr_t res;
            res = esp_netconn_connect(ncs[i], "1.2.3.4", 12345 ); /* sic! */
            printf("esp_netconn_connect(ncs[%i]) result: %d\r\n", (int)i, (int)res);
        } else {
            printf("ncs[%i] == NULL\r\n", (int)i);
        }
    }
```

Results:
```
esp_netconn_connect(ncs[0]) result: 0
esp_netconn_connect(ncs[1]) result: 0
esp_netconn_connect(ncs[2]) result: 0
esp_netconn_connect(ncs[3]) result: 0
esp_netconn_connect(ncs[4]) result: 0
esp_netconn_connect(ncs[5]) result: 0
esp_netconn_connect(ncs[6]) result: 0
esp_netconn_connect(ncs[7]) result: 0
esp_netconn_connect(ncs[8]) result: 0
esp_netconn_connect(ncs[9]) result: 0
```

UART log:
```
OK
AT+CIPSTART=4,"TCP","1.2.3.4",12345

ERROR
4,CLOSED
AT+CIPSTATUS
STATUS:4

[ 10 times ]
```

ev-ermakov, 2019-02-03:
The problem is that the server does not exist, but there is no error.

MaJerle, 2019-02-03:
Please make sure you have latest version, commited less than 1h ago. Here my log on PC (I combine UART + printf logs).

```
AT+CIPSTATUS
STATUS:2

OK
AT+CIPSTART=4,"TCP","example.com",80
+LINK_CONN:0,4,"TCP",0,"93.184.216.34",80,8052

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[0]) result: 0 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
AT+CIPSTART=3,"TCP","example.com",80
+LINK_CONN:0,3,"TCP",0,"93.184.216.34",80,37345

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[1]) result: 0 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
AT+CIPSTART=2,"TCP","example.com",80
+LINK_CONN:0,2,"TCP",0,"93.184.216.34",80,23897

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[2]) result: 0 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
AT+CIPSTART=1,"TCP","example.com",80
+LINK_CONN:0,1,"TCP",0,"93.184.216.34",80,40000

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[3]) result: 0 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
AT+CIPSTART=0,"TCP","example.com",80
+LINK_CONN:0,0,"TCP",0,"93.184.216.34",80,41773

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","93.184.216.34",80,41773,0
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[4]) result: 0 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","93.184.216.34",80,41773,0
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[5]) result: 10 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","93.184.216.34",80,41773,0
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[6]) result: 10 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","93.184.216.34",80,41773,0
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[7]) result: 10 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","93.184.216.34",80,41773,0
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[8]) result: 10 <---------------------
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","93.184.216.34",80,41773,0
+CIPSTATUS:1,"TCP","93.184.216.34",80,40000,0
+CIPSTATUS:2,"TCP","93.184.216.34",80,23897,0
+CIPSTATUS:3,"TCP","93.184.216.34",80,37345,0
+CIPSTATUS:4,"TCP","93.184.216.34",80,8052,0

OK
esp_netconn_connect(ncs[9]) result: 10 <---------------------
```

ev-ermakov, 2019-02-03:
Yes, I have already updated the library.

But the thing is that I was trying to connect to a non-existent server.

Now try to connect to 1.2.3.4:12345, for instance.

MaJerle, 2019-02-03:
I see now, ok confirmed from my side aswell.

ev-ermakov, 2019-02-03:
Yes, because you are trying to connect to an existing server (example.com).

But the thing is that I was trying to connect to a non-existent server.

Try to connect to 1.2.3.4:12345, for instance.

```c
        res = esp_netconn_connect(ncs[i], "1.2.3.4", 12345 );
```

And you still get `res = 0`.