MaJerle/lwesp #37: esp_netconn_receive() returns not all data

ev-ermakov, 2019-03-01:
Hi.

Suppose we have following setup. A TCP server that sends N bytes at the maximum possible bit rate immediately after the client connects.

Server:
```console
$ wc -c < SOMEFILE
9504
$ socat tcp4-listen:8888,forever,fork,reuseaddr file:SOMEFILE
```

And a client:
```c
    esp_pbuf_p buf;
    esp_netconn_p conn;
    size_t N = 9504;
    size_t n_recv;
    size_t n_recv_total;
    espr_t status;

    conn = esp_netconn_new(ESP_NETCONN_TYPE_TCP);
    assert(conn != NULL);

    if ((status = esp_netconn_connect(conn, "192.168.1.1", 8888)) != espOK)
    {
        printf("esp_netconn_connect() failed (%u)", status);
    }
    else
    {
        n_recv_total = 0;

        for (;;)
        {
            if ((status = esp_netconn_receive(conn, &buf)) != espOK)
            {
                printf("Expected: %u, received: %u", N, n_recv_total);
                break;
            }

            n_recv = esp_pbuf_length(buf, 1);
            n_recv_total += n_recv;

            esp_pbuf_free(buf);
            buf = NULL;
        }

        if (status != espCLOSED)
            esp_netconn_close(conn);

        esp_netconn_delete(conn);
    }
```

And here is what the client prints:
```console
Expected: 9504, received: 8760
```

To reproduce this problem, make sure the network latency is minimal.

MaJerle, 2019-03-01:
Can you show me UART log from ESP8266? I have to make sure that ESP outputs all data as I know it may fail sometimes.

ev-ermakov, 2019-03-01:
Okay, but a bit later.

MaJerle, 2019-03-01:
Sure. In any case. If you received espCLOSED return from receive function, then it means esp stack properly handles situation.

Then there are 2 options:
- ESP did not send all data, may happen and I've seen this already
- There was no memory in stack for last packet. Less likely as you read data pretty fast.

ev-ermakov, 2019-03-01:
Yeah, you were right about ESP. It seems that this is a bug in the module :(

closed 2019-03-01 by ev-ermakov

MaJerle, 2019-03-01:
Issue has been reported many times to espressif, but without resolution.

ev-ermakov, 2019-03-01:
Well, I will look for workarounds.