MaJerle/lwesp #38: how to port to other platform.

xelll, 2019-03-21:
i am so sorry about that, i am not so clear about how to port to other platform. i see some template in system dir. but i can only find how send data to esp8266 module. and i do not know how to recveive data from esp8266.

MaJerle, 2019-03-21:
Receiving of the data (UART) must be resolved with care on your side. You may use interrupts or (like all examples here), use separate thread and DMA to receive data. Library does not care about how you will implement it.

xelll, 2019-03-21:
https://github.com/MaJerle/ESP_AT_Lib/blob/5e46e1ad14bdb4bb190be5899da109a1ec86ee9d/src/system/esp_ll_win32.c#L155
thanks , i think i can use this to input data.

closed 2019-03-21 by xelll