MaJerle/lwesp #41: Full re-init esp_ll in case of system errors

biovoid, 2019-04-23:
I've tested with esp_reset, but I realize that it does not reset chip. I mean RST command is ok, and may be sometimes its useful. But in case of some hardware errors its better to have reset with safe closing any socket and fully re-init harware esp_ll layer. For example, if there is on-board power problem, and esp was reset AND libraary use not default baudrate then there is no chance to restore communication.

_Originally posted by @biovoid in https://github.com/MaJerle/ESP_AT_Lib/issues/39#issuecomment-485877306_

biovoid, 2019-04-23:
List of item to be handled in this case:
1. Sockets should close
2. ESP_EVT_WIFI_DISCONNECTED should raise if we were connected
3. Reinit latest used baudrate/mode setting
4. Any new API access can be blocked until module restarts

MaJerle, 2019-04-23:
First 2 steps of your proposal are already working. When there is reset detection (or if you manually call `esp_set_device_present` it will manually reset things for upper application layer.

3. To get default baudrate, we need to implement one more command during reset sequence to know what is baudrate after module starts and react accordingly. Here the question arises, how to detect module reset (or issues) if current baudrate setting is different than the one after reset?

For now I propose you to do this:
- When you realize there is an error with module, call `esp_set_device_present` function informing stack that device is not connected anymore.
- Manually reset device, any data received from ESP to MCU will be ignored as stack has flag "no device" set to `1`
- Then set back `esp_set_device_present` back to `1` and reset sequence will proceed accordingly.

I see your point and it seems valid, but then we have to make sure user cannot change default UART setting. Default UART setting is then `ESP_CFG_AT_PORT_BAUDRATE` macro and if there is reset detected (btw, how to detect reset if current UART is at higher speed than ESP after reset), stack manually calls init function to change the baudrate back to default. Same goes in case user calls `esp_set_device_present`. Later, user can increase baudrate back to the one before reset.

I can implement new callback for reset in the `esp_ll_t` structure. If it is not default value, callback is called for setting GPIO state instead of sending command, or using command if callback is set to NULL.

MaJerle, 2019-04-23:
See commits https://github.com/MaJerle/ESP_AT_Lib/commit/75ca7e4806ffedd8c2e3abf313c145f1922685e7 and https://github.com/MaJerle/ESP_AT_Lib/commit/6d4b677a6efc45ff5366c29a05a9e88fd543a8ae. Now if you modify baudrate for some reason to higher, it is not possible by the UART protocol, to detect reset of the device. If, however, you can determine that there was an issue with device, you may call `esp_reset` function which will reset ESP with GPIO and also set baudrate back to default before releasing GPIO state.

biovoid, 2019-04-24:
I've tested your latest commit, it looks stable for now. I am calling esp_reset when ESP_EVT_CMD_TIMEOUT occurs and everething is restarted properly. I will test it more.

MaJerle, 2019-04-30:
I have update the lib. Now if you plan to use `esp_reset`, it will first reset entire stack, including baudrate back to ESP default one. So take care that you use actual reset pin in case of trying to reset ESP device when changing baudrate settings.

This is at the end the most stable solution.

MaJerle, 2019-04-30:
To give you an idea how to handle baudrate change. You may do it in reset event in your custom callback function:

```c
espr_t
esp_callback_func(esp_evt_t* evt) {
    switch (esp_evt_get_type(evt)) {
        case ESP_EVT_RESET: {
            /* Check if reset is OK */
            if (esp_evt_reset_get_result(evt) == espOK) {
                esp_set_at_baudrate(921600, NULL, NULL, 0);
            }
            break;
        }
        case ESP_EVT_CMD_TIMEOUT: {
            esp_reset(NULL, NULL, 0);       /* Call reset function */
            break;
        }
        default: break;
    }
    return espOK;
}
```

> Be advised that your reset needs to include physical pin and reset callback on low-level part.

closed 2019-04-30 by MaJerle