MaJerle/lwesp #44: ESP_32 support

Masi2017, 2019-08-01:
Hi All,

I was wondering if library supports esp_32 WROOM? I tried to set the ESP_CFG_ESP32 to 1 and  ESP_CFG_ESP8266 to 0 to be able to use the library for ESP32 but I've received an error at the compile time. In esp_config_default.h it's mentioned that ESP_CFG_ESP32 must be set to 0, so does this mean that library doesn't support ESP32?

MaJerle, 2019-08-01:
For now yes.

closed 2019-08-01 by MaJerle

Masi2017, 2019-08-01:
Hi Sorry I didn't get that what do you mean by yes? It does support? But I can't enable that.

MaJerle, 2019-08-01:
For now it does not support.

Masi2017, 2019-08-02:
OK thanks

MaJerle, 2019-08-02:
Informarion I have is that Espressif thinks to made ESP8266 and ESP32 comnands compatible. So far this is not the case, some commands accept different parameters, like CWDHCP, for example.

Once migration is done, library will support both. Put yourself to watch list and you will be notified.