MaJerle/lwesp #47: MQTT decoding issue

Masi2017, 2019-10-03:
Hi @MaJerle 
Thanks for all your efforts.
I found an issue in  mqtt_parse_incoming function when data length is >=128
I modified the function and fixed it as bellow :
`
static uint8_t
mqtt_parse_incoming(esp_mqtt_client_p client, esp_pbuf_p pbuf) {
    size_t idx, buff_len = 0, buff_offset = 0;
    const uint8_t* d;
    uint8_t ch;
    uint8_t multiplier = 1;

    do {
        buff_offset += buff_len;                /* Calculate new offset of buffer */
        d = esp_pbuf_get_linear_addr(pbuf, buff_offset, &buff_len); /* Get address pointer */

        idx = 0;
        while (d != NULL && idx < buff_len) {   /* Process entire linear buffer */
            ch = d[idx++];                      /* Get element */
            switch (client->parser_state) {     /* Check parser state */
                case MQTT_PARSER_STATE_INIT: {  /* We are waiting for start byte and packet type */
                    ESP_DEBUGF(ESP_CFG_DBG_MQTT_STATE,
                        "[MQTT] Parser init state, received first byte of packet 0x%02X\r\n", (unsigned)ch);

                    /* Save other info about message */
                    client->msg_hdr_byte = ch;  /* Save first entry */
                    client->msg_rem_len = 0;    /* Reset remaining length */
                    client->msg_curr_pos = 0;   /* Reset current buffer write pointer */

                    client->parser_state = MQTT_PARSER_STATE_CALC_REM_LEN;
                    break;
                }
                case MQTT_PARSER_STATE_CALC_REM_LEN: {  /* Calculate remaining length of packet */
                  client->msg_rem_len += (ch & 127) * multiplier;
                   multiplier *= 128;
                //    client->msg_rem_len <<= 7;  /* Shift remaining length by 7 bits */
                //    client->msg_rem_len |= (ch & 0x7F);
                    /* TODO: Ignore too-big messages */
                    if (!(ch & 0x80)) {         /* Is this last entry? */
                        ESP_DEBUGF(ESP_CFG_DBG_MQTT_STATE,
                            "[MQTT] Remaining length received: %d bytes\r\n", (int)client->msg_rem_len);

                        if (client->msg_rem_len > 0) {
                            client->parser_state = MQTT_PARSER_STATE_READ_REM;
                        } else {
                            mqtt_process_incoming_message(client);
                            client->parser_state = MQTT_PARSER_STATE_INIT;
                        }
                    }
                    break;
                }
                case MQTT_PARSER_STATE_READ_REM: {  /* Read remaining bytes and write to RX buffer */
                    client->rx_buff[client->msg_curr_pos++] = ch;   /* Write received character */

                    if (client->msg_curr_pos == client->msg_rem_len) {
                        ESP_DEBUGF(ESP_CFG_DBG_MQTT_STATE,
                            "[MQTT] Packet parsed and ready for processing\r\n");

                        mqtt_process_incoming_message(client);  /* Process incoming packet */
                        client->parser_state = MQTT_PARSER_STATE_INIT;  /* Go to initial state and listen for next received packet */
                    }
                    break;
                }
                default:
                    client->parser_state = MQTT_PARSER_STATE_INIT;
            }
        }
    } while (buff_len);
    return 0;
}
`

MaJerle, 2019-10-03:
Thanks. This is not perfect solution. If packet is separated at length level, modifier will be again 1 as it is local variable.

What exactly is wrong with my implementation please?

Masi2017, 2019-10-03:
Yes it's true, how about adding modifier field to esp_mqtt_client_t?

MaJerle, 2019-10-03:
Done. Thanks for reporting.

closed 2019-10-03 by MaJerle

MaJerle, 2019-10-04:
One more feature will come to MQTT. When received packet is within single pbuf [not fragmented], it is possible to manipulate it with zero-copy hence improved performance of the system.