MaJerle/lwesp #48: Callback when station connects to SoftAP on ESP8266

tvandergeer, 2019-10-26:
For the last couple of days I've been working on an integration between a STMF769i-discovery and a ESP8266 WiFi module using your library. 

A long story short: i'm able to successfully initialise the library and instruct the ESP8266 to setup an access point (with WPA_WPA2). Very similar to the [Win32 access_point_rtos example](https://github.com/MaJerle/ESP_AT_Lib/blob/develop/examples/win32/access_point_rtos/main.c#L57). Once I connect my phone to the access point the callback for the event ESP_EVT_AP_CONNECTED_STA is not fired. Is this a known issue?

A bit more about my setup:
I've got several ESP8266 lying around:
- A Wemos D1 mini pro (v1.0.0) with 16MB
- A Lolin new NodeMCU v3 with 4MB
- ESP-01 module with 1MB

I've successfully updated the Wemos and Lolin to the stock "ESP8266 IDF AT Bin V2.0" firmware from the espressif website. The ESP-01 only has 1MB, which is not large enough for the stock firmware. Building my own firmware for the smaller memory has not yet been successful. For 2MB+ I was successful.

I've put a working setup together using the stock 2.0 firmware on the NodeMCU and connecting the STM32F769i discovery with a handful of breadboard cables. I had to use the UART2 ports on the NodeMCU for UART TX/RX. Took me a while to figure that out. Only to find out later that you've (kind of) mentioned this [in this post](https://github.com/MaJerle/ESP_AT_Lib/issues/40#issuecomment-520260634) as well. Or it's actually not the UART2 port, but the swapped RTS0/CTS0 pins for UART0. Espressif acutally mentions this [here](https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md) and how you can swap this back when building your own firmware.

The ESP_AT_Lib is initialising correctly. This is the output from the serial console

```
Initializing ESP-AT Lib
ESP callback evt: 0
Library initialized!
ESP callback evt: 1
Device reset detected!
ESP callback evt: 4
[THREAD] Timeout in produce thread waiting for command to finish in process thread
ESP callback evt: 3
ESP callback evt: 1
Device reset detected!
ESP callback evt: 2
ESP-AT Lib initialized!
```

The "ESP callback evt: N" lines are from an added `printf("ESP callback evt: %d\r\n", esp_evt_get_type(evt));` line in esp_callback_func.

What I did notice it that the ESP8266 on the v2.0 firmware is no longer sending the following lines on the UART once a new station connects:
```
+STA_CONNECTED:"xx:xx:xx:xx:xx:xx"
+DIST_STA_IP:"xx:xx:xx:xx:xx:xx","192.168.4.2"
```
It was doing that with the older firmware. This seems to be related to my issue with the missing events.

tvandergeer, 2019-10-26:
FYI. This is what I'm reading on GPIO15 (TXD2, aka the swapped TXD0 pin from the stock 2.0 firmware)

```
ready
ATE1

OK
AT+SYSMSG=3

OK
AT+GMR
AT version:2.0.0.0(88c2cde - Jul  9 2019 11:03:33)
SDK version:v3.2
compile time:Jul 15 2019 09:23:26
Bin version:2.0.0(WROOM-02)

OK
AT+CWMODE=3

OK
AT+CWDHCP?
+CWDHCP:3
OK
AT+CIPMUX=1

OK
AT+CWLAPOPT=1,31

OK
AT+CIPSTATUS
STATUS:1

OK
AT+CIPAP?
+CIPAP:ip:”192.168.4.1”
+CIPAP:gateway:”192.168.4.1”
+CIPAP:netmask:”255.255.255.0”

OK
AT+CIPAPMAC?
+CIPAPMAC:” xx:xx:xx:xx:xx:xx”

OK
AT+CIPDINFO=1

OK
AT+CWMODE=2

OK
AT+CIPAP?
+CIPAP:ip:”192.168.4.1”
+CIPAP:gateway:”192.168.4.1”
+CIPAP:netmask:”255.255.255.0”

OK
AT+CWDHCP?
+CWDHCP:3
OK
AT+CIPAPMAC?
+CIPAPMAC:” xx:xx:xx:xx:xx:xx”

OK
AT+CWSAP=“myapname”,”myappassword”,9,4,5,0

OK
```
Once I connect a station (e.g. my phone) via WiFi there's no additional data.

MaJerle, 2019-10-26:
@tvandergeer I do confirm that ESP8266 AT v2.0 does not output anything once new station is connected to access point hence no callback function.

AT v2.0 is derived from ESP32-AT architecture and this is missing (for now) there.

This is official repository, where Espressif is developing new features: https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md

If you report something, let me know.

Cheers, Tilen

tvandergeer, 2019-10-26:
Thanks for your quick response!

I will report this in the esp-at repo. Meanwhile, do you suggest that I go back to 1.6.0 firmware and use a version of your esp_at_lib from a few months back? Then I can use my ESP-01 as well.

Or should I poll for newly connected stations? 

Do you know of any other “problems” with the V2.0 firmware that I should take into account?

MaJerle, 2019-10-27:
> I will report this in the esp-at repo. Meanwhile, do you suggest that I go back to 1.6.0 firmware and use a version of your esp_at_lib from a few months back? Then I can use my ESP-01 as well.

No, use at least 1.7 as there were many performance improvements introduced and stability of firmware.

> Or should I poll for newly connected stations?

That's one way, definitely. How soon do you need to detect new station? Can it be with 10-20sec latency?

> Do you know of any other “problems” with the V2.0 firmware that I should take into account?

I've noticed problems with manual TCP receive data on 2.0, but this is going to be fixed in next release (confirmed & tested beta firmware already).

tvandergeer, 2019-10-28:
> If you report something, let me know.

Here it is: https://github.com/espressif/esp-at/issues/252

closed 2019-11-18 by MaJerle