MaJerle/lwesp #50: Creation of server fails because of missing parameter

tvandergeer, 2019-10-28:
I've found an issue when trying out the netconn API. Setting up a new server on port 80 generated (and sent) the following AT command sequence:

```
AT+CIPSERVERMAXCONN=5

OK
AT+CIPSERVER=,80

ERROR
```
Notice the ERROR response on the last command. It's missing the first paramter of the AT+CIPSERVER command. This should be either a 0 (delete server) or 1 (create server). It looks like this can be fixed by adding the `AT_PORT_SEND_CONST_STR("1");`. This is the full code block:

```
case ESP_CMD_TCPIP_CIPSERVER: {         /* Enable or disable server */
            AT_PORT_SEND_BEGIN_AT();
            AT_PORT_SEND_CONST_STR("+CIPSERVER=");
            if (msg->msg.tcpip_server.en) {     /* Do we want to enable server? */
                AT_PORT_SEND_CONST_STR("1"); // this line is added
                espi_send_port(msg->msg.tcpip_server.port, 0, 1);
            } else {                            /* Disable server */
                AT_PORT_SEND_CONST_STR("0");
            }
            AT_PORT_SEND_END_AT();
            break;
        }

```

tvandergeer, 2019-10-28:
This is a patch
[patch.diff.txt](https://github.com/MaJerle/ESP_AT_Lib/files/3779593/patch.diff.txt)

MaJerle, 2019-10-28:
Im aware of these issues.
You should use master branch, not develop.

closed 2019-10-28 by MaJerle