MaJerle/lwesp #51: Issue with CIPCLOSE

Masi2018, 2019-10-31:
In my applications sometimes it happens although MQTT port close command is sent ( AT+CIPCLOSE) , no response (either CLOSED or ERROR ) is received and since the MQTT connection state is ESP_MQTT_CONN_DISCONNECTING ,  ESP AT Lib stays in this state forever no matter if esp_mqtt_client_disconnect is sent again because it never try to send AT+CIPCLOSE.

MaJerle, 2019-10-31:
Does this mean issue is with ESP firmware on Espressif side? If there is no x,CLOSED event, it is tough to do much.

Masi2018, 2019-10-31:
sometimes ESP firmware in Espressif side doesn't respond back to AT+CIPCLOSE so ESP AT LIB stays in ESP_MQTT_CONN_DISCONNECTING forever.

Masi2018, 2019-10-31:
I added the functions below to force sending AT+CLIPCLOSE again even if is in ESP_MQTT_CONN_DISCONNECTING state.

espr_t mqtt_force_close(esp_mqtt_client_p client) {
    espr_t res = espERR;
     {
        res = esp_conn_force_close(client->conn, 0);  /* Close the connection in non-blocking mode */
        if (res == espOK) {
            client->conn_state = ESP_MQTT_CONN_DISCONNECTING;
        }
    }
    return res;
}

espr_t
esp_conn_force_close(esp_conn_p conn, const uint32_t blocking) {
    espr_t res;
    ESP_MSG_VAR_DEFINE(msg);

    ESP_ASSERT("conn != NULL", conn != NULL);

    /* Proceed with close event at this point! */
    ESP_MSG_VAR_ALLOC(msg, blocking);
    ESP_MSG_VAR_REF(msg).cmd_def = ESP_CMD_TCPIP_CIPCLOSE;
    ESP_MSG_VAR_REF(msg).msg.conn_close.conn = conn;
    ESP_MSG_VAR_REF(msg).msg.conn_close.val_id = espi_conn_get_val_id(conn);

    flush_buff(conn);                           /* First flush buffer */
    res = espi_send_msg_to_producer_mbox(&ESP_MSG_VAR_REF(msg), espi_initiate_cmd, 1000);

    if (res == espOK && !blocking) {            /* Function succedded in non-blocking mode */
        esp_core_lock();
        ESP_DEBUGF(ESP_CFG_DBG_CONN | ESP_DBG_TYPE_TRACE,
            "[CONN] Connection %d set to closing state\r\n", (int)conn->num);
        conn->status.f.in_closing = 1;          /* Connection is in closing mode but not yet closed */
        esp_core_unlock();
    }
    return res;
}

Masi2018, 2019-10-31:
Here is an screenshot when ESP32 doesn't reply back to AT+CIPCLOSE ( +CLOSED is not sent) so MQTT connection state remains in ESP_MQTT_CONN_DISCONNECTING  state conn->status.f.in_closing stays in in_closing .

[image]

MaJerle, 2019-10-31:
This is defjnitely not a solution. We need to report to esp-at Espressif Github.