MaJerle/lwesp #52: Feature request: support for SSL client certificates

tvandergeer, 2019-11-22:
The esp-at firmware supports using client certificates for SSL connections. (See [here](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#cmd-SSLCCONF)). This is required when setting up an SSL link to e.g. AWS IOT. I have successfully tested the concept by providing manual AT commands. Steps to reproduce:

1. make sure to have client certificate/key/CA-cert on the ESP. (I chose them to be at index 0)
2. init ESP as usual and make connection to internet
3. configure an NTP server to get current time and date (required to provide/validate SSL certificates) using `AT+CIPSNTPCFG`
4. Configure SSL using `AT+CIPSSLCCONF` and specify auth_mode 3 and certificate indexes.
5. Start SSL connection using `AT+CIPSTART`

For a full UART log of my manual PoC see [this gist](https://gist.github.com/tvandergeer/da1e753643b7420ecddf086fac6ea893) (redacted sensitive info)

Step 3. is not complex. It's just another init step which can be solved with some config (timezone and NTP host) and a new command: `AT+CIPSNTPCFG`
Step 4. is a bit more complex (I think), because the `AT+CIPSSLCCONF` requires the `<link ID>` of the connection. This value is not known prior to the `AT+CIPSTART` command. I guess it's possible to generate this value in the `AT+CIPSSLCCONF` command execution and store it in the connection object and check in the `AT+CIPSTART` command whether this value has previously been set (and the connection is in the correct state). Or do you see any other possibility?

Do you accept pull requests? First for step 3 and when we figured out an approach for Step 4 I can work on that as well.

MaJerle, 2019-11-22:
My next focus is to implement AT+CIPSTARTEX where you do not need to specify connection ID. This will decrease chances for race conditions when server is also enabled. But needs some tests on ESP side first as it is not yet stable as I would expect to be.

Effectively this is contradictive with how SSLCCONF works. One wants connection ID while second supports without it.

Unless Espressif believes when you use SSL client, you are not interested in server mode.

Anyway, SNTP should not be a problem. Actually it already exists. What we can do is to add option to set default NTP server in config and this is then used during init sequence.

tvandergeer, 2019-11-25:
Ok, I see. Using `AT+CIPSTARTEX` is contradictive to using `AT+CIPSSLCCONF`. Another option is to just add `AT+CIPSSLCCONF` functionality to the library and allowing the user to specify the link ID. When the Max connection value is specified for the Link ID it will use the configuration on all SSL connections. Would you accept a pull request for an `esp_ssl_configure` method?

And you're right on the NTP functionality. It's already available. For me there's no need to make this part of the init sequence.

MaJerle, 2019-11-25:
If you ask me, Espressif should separate AT+CIPSTART command for SSL only, where all the setup of AT+CIPSSLCCONF would be included in single call. Something like AT+CIPSTARTSSLEX. Single command with host/port/ssl_config setup.

I will accept pull request, but it must strictly comply with coding rules already applied to the lib.

tvandergeer, 2019-11-26:
Thanks for merging the pull request! I will close this comment.

closed 2019-11-26 by tvandergeer