MaJerle/lwesp #54: Workaround for missing +IDP message when server is in passive mode

tvandergeer, 2019-12-06:
There's an issue in the esp-at firmware where no `+IDP` message is received when in "passive mode" (Manual TCP receive) and a server is listening for connections. I've mentioned this issue here: https://github.com/MaJerle/esp-at-lib/issues/46

The issue report for the esp-at firmware is here: https://github.com/espressif/esp-at/issues/294

We can (probably) fix/workaround this in the library as well by sending a `AT+CIPRECVLEN?` right after receiving a `+LINK_CONN` when it's a server connection and in "passive mode". Would you be inclined for this?

MaJerle, 2019-12-06:
As per my understanding, you would issue `AT+CIPRECVLEN?` command when connection is opened as server?

This can be done. Did you verify if it increases stability? Anyway this is again ugly workaround :)

tvandergeer, 2019-12-06:
Yes, when a `+LiNK_CONN` message is received from a new incoming client connection. Not when you create a new server with `AT+CIPSERVER=1,80`.

I know; it's ugly. I will check for stability improvement.

MaJerle, 2019-12-08:
I have added manual check just after connection becomes active. This is valid for client or server modes.

https://github.com/MaJerle/esp-at-lib/commit/c8dda69f6db8fe4682cf0400b2523ed30bddfb5e

This is a one-time check and is not regular with some X intervals. Maybe it would be good to implement manual length check at every X seconds.

Let me know.

tvandergeer, 2019-12-08:
Ah, that looks good! I've tested a very similar approach where I reused the `espi_conn_manual_tcp_try_read_data(conn);` method. But I had to 'trick' it into thinking there was data in the buffer. So at line 878 in esp_int.c I had the following code:
```
#if ESP_CFG_CONN_MANUAL_TCP_RECEIVE
                /* See esp-at issue: https://github.com/espressif/esp-at/issues/294 */
                if(conn->status.f.client == 0){ /* conn is a server connection */
                    conn->tcp_available_bytes = 1; /* let's assume there is some data that we can fetch from the new connection */
                	espi_conn_manual_tcp_try_read_data(conn);
                }
#endif /* ESP_CFG_CONN_MANUAL_TCP_RECEIVE */
```
Not sure whether it should be limited to server connections.

I like your solution better, because it opens the flexibility to use elsewhere as well. Will give that a try.

tvandergeer, 2020-05-04:
Looks like they fixed this issue in the esp-at firmware on Master

closed 2021-03-30 by MaJerle