MaJerle/lwesp #56: ESP8266 returns busy p...

Szbuli, 2020-02-16:
I am using F407 with ESP8266. I upgraded the firmware of the ESP8266 to latest version:

> AT version:1.6.2.0(Apr 13 2018 11:10:59)
> SDK version:2.2.1(6ab97e9)
> compile time:Jun  7 2018 19:34:26

The ESP8266 most of the time returns: 

> busy p...

If i put a few ms sleep in the `send_data` function the library keeps reseting the ESP8266.

If you need more messages between the F407 and the ESP8266 let me know.

MaJerle, 2020-02-16:
Minimal AT version is 2.1.0.

closed 2020-02-16 by MaJerle

Szbuli, 2020-02-16:
I have installed ESP8266 NONOS SDK V2.2.1.

MaJerle, 2020-02-16:
> I have installed ESP8266 NONOS SDK V2.2.1.

We are talking about AT version, not SDK version.

Szbuli, 2020-02-16:
On the offical site the latest AT version which i could find is V1.7.2, released at 2020.01.20.

Could you help me where can i download a sufficient version?

MaJerle, 2020-02-16:
https://www.espressif.com/en/products/hardware/esp8266ex/resources

Under At section.

Szbuli, 2020-02-16:
I was using that page for downloading the firmware. The highest version's are 

> ESP8266 NonOS AT Bin V1.7.2 - 2020.01.20
> ESP8266 IDF AT Bin V2.0 - 2019.07.16

There are no versions that are higher than 2.0

MaJerle, 2020-02-16:
2.0 is ok