MaJerle/lwesp #57: POSIX (pthreads) backend?

laanwj, 2020-02-18:
It would be nice to have a POSIX backend for this library, to use with embedded Linux on boards with a ESP8285 (for ex. https://github.com/damien-lemoal/riscv64-nommu-buildroot).

I might have a stab at this.

MaJerle, 2020-02-18:
You may adapt SYS layer to fit to posix implementation, but ESP SYS layers stays the same.

franccan, 2021-08-09:
Are there any plans to add support to posix compliant OS (i.e. lwesp_sys_posix.c)?

I think this could be an easy way to add MCU capabilities to an embedded linux device. This is because the "Driver AT Commands" support driving I2S, SPI, ADC and PWM peripherals.
https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Driver_AT_Commands.html

closed 2022-04-14 by laanwj