MaJerle/lwesp #60: Performans

yldzmuhammed, 2020-03-30:
Hi @MaJerle,

Great thanks for such beatifull library.

I have some performance concern. Have you done any performance test?

Is it possible to make a http camera server with this library?

Stm32 reads camera data over dcmi, serves via esp32 over uart. Is it possible, can uart speed overcome this?

MaJerle, 2020-03-30:
This is not library performance concern, it is UART concern then.
That needs to be evaluated on your side.

STM32 will handle data faster than UART can do it, I'm sure about that.

yldzmuhammed, 2020-03-30:
In fact at this point we are sharing same ideas. STM32 definatly can handle faster than uart. But will client side effect from it? I mean when client connect to the server, will it be smooth or will it freeze most of the time? What do you think?

MaJerle, 2020-03-30:
This I cannot say. Here library has unfortunately nothing to do. I would think but I dont know how to make lib more attractive from performance point of view.

You have to calculate FPS, resolution and UART speed.

MaJerle, 2020-03-30:
Sorry, this is not the right place to calculate it. It is anyway, 800*480*2*30 bytes per second.

yldzmuhammed, 2020-03-31:
You are right. Sorry to disturb..

closed 2020-03-31 by yldzmuhammed