MaJerle/lwesp #63: Support for AT+SYSFLASH?

BertLammers, 2020-04-17:
First, thanks for this fantastic library!

I am currently working on a project where I need to update SSL client certificates. This can be done using the AT+SYSFLASH command (ESP32 only). Are there any plans to introduce support for this command?

If not I will consider writing it myself.

MaJerle, 2020-04-18:
Thanks Bert for your interest.
I have it in my roadmap but no timeline yet.

sicris, 2020-12-29:
@BertLammers... I have the same use-case.  I need to update my certificates.
Do you have any patch that I can start with?

sicrisembay, 2020-12-30:
I did a rough initial implementation for AT+SYSFLASH.
Tested Operations: Erase, Write and Read
To be implemented: Parsing of AT+SYSFLASH? query

Pull request will follow after cleaning-up the code.
Forked repo: https://github.com/sicrisembay/lwesp.git
Branch: dev_sysflash

BertLammers, 2023-03-17:
@sicrisembay 2 years later but using your AT+SYSFLASH implementation now, works perfectly! ;)

MaJerle, 2023-10-01:
This is now added, btw.