MaJerle/lwesp #64: Feature proposal: Support for SDIO in stead of UART communication

tvandergeer, 2020-05-05:
Please see this question (not mine) on the esp-at issues list: https://github.com/espressif/esp-at/issues/379

For my use case I'm downloading large files (>10MB) and even using a relatively high UART clock rate of 2.000.000 this takes a few minutes. (My assumption is that the clock rate of the UART is the bottleneck here)

I am exploring possibilities to improve the performance. So after reading the above issue and reading up on the [low level driver documentation for your library]( https://docs.majerle.eu/projects/esp-at-lib/en/latest/user-manual/porting-guide.html#implement-low-level-driver) I thought it should be feasible.

Do you have any thoughts on this? Did you ever try this? My target platform is an STM32 MCU.

NB. After writing this message I found that the same user that posted the above question to the esp-at issues also send in an issue for this project: https://github.com/MaJerle/esp-at-lib/issues/60 It looks like he's attempting to do the same...

MaJerle, 2020-05-06:
This is fully valid request. I'm considering it for some time now but I have high lack of time to implement this. Next feature that will be supported is built-in MQTT and AT over SPI.

tvandergeer, 2020-05-09:
Ah! I was not aware that AT over SPI was an option too. But I researched it and found the “AT through HSPI” option. However it’s not showing up in make menuconfig. I suspect that this is because I’m using an ESP32 and this feature is limited to ESP8266, but I’m not sure. The Readme file [here](https://github.com/espressif/esp-at/tree/master/main/interface/hspi) specifically names it “ESP8266 AT through SPI” but that might just be an inconsistency in the documentation. 

In [thisremark](https://github.com/espressif/esp-at/issues/379#issuecomment-605621120) one of the esp-at project contributors suggests to use SDIO over SPI. 

I’m considering to investigate this a bit further and play around with the SDIO example where 2 ESP32’s are connected via SDIO. Based on the results and what I learn from this I might start on an STM32 port.

caowent, 2020-07-01:
> Ah! I was not aware that AT over SPI was an option too. But I researched it and found the “AT through HSPI” option. However it’s not showing up in `make menuconfig`. I suspect that this is because [...]

[host at spi](https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host)
This is an example of AT on the host side (STM32). The MCU works on the spi and the ESP32 works on the SPI mode of SDIO.
The TCP throughput is about 1.2MBytes/s.