MaJerle/lwesp #68: comment misleading

jerrytry1, 2020-08-04:
Hi, In your code lwesp_int.c    line 738-739 and   748-749
<code>
if (!CMD_IS_CUR(LWESP_CMD_WIFI_CWJAP)) {  /* ### **In case of auto connection** */
     lwesp_sta_getip(NULL, NULL, NULL, NULL, NULL, 0);  /* Get new IP address */
</code>
Actually, it is not only in case of auto connection.   Whenever I call "lwesp_sta_join()",  this function "lwesp_sta_getip" will be triggered, and this is where I got my IP address. Right?

the comment should be something like "In case of receive "WIFI CONNECTED".

And I suggest to remove line 738-739, and keep 748-749  sometimes,  getip is 0.0.0.0 at this point. 

Thanks.

MaJerle, 2020-08-04:
Like 738, ip address check is called if current command is not join. If it is, then line 1454 is called.

jerrytry1, 2020-08-04:
Thanks for quick reply. it is very nice library that you did. Appreciated.

closed 2020-08-04 by jerrytry1