MaJerle/lwesp #74: HAL Support ?

Chleba, 2021-03-03:
Will there be support for HAL Drivers ?

MaJerle, 2021-03-04:
So far I didnt seeva need for it. Demo based on LL for stm32 is not enough?

Chleba, 2021-03-04:
I was just asking if you have it somehow prepared or something. I will make a PR later for you, if you want to have it as a part of Your library. Thanks :+1:

MaJerle, 2021-03-04:
Don't close it so quickly :)

Having HAL makes sense only if we can implement DMA receive especially with high traffic. Latest versions of HAL (H7, G4, L5 + some others) support DMA receive with unknown length on RX side. IF you can implement such thing with HAL and make a pull request to develop branch, I will merge it.

Chleba, 2021-03-04:
I am sorry about that - closing this "Issue" too soon. I will let you know. Thank you.

jerrytry1, 2021-03-05:
That is my current job.  add HAL DMA support.  with stm32xxxxx

closed 2021-03-30 by MaJerle