MaJerle/lwesp #84: Feature proposal: wrap up lwesp_input/lwesp_input_process

niedong, 2021-05-17:
Hi, I think it would be nice for lwesp library to have another thread to wrap up `lwesp_input/lwesp_input_process` for processing incoming data, and provide user with `receive_byte/receive_range` function to use in interrupt callbacks/DMA callbacks, etc. Or is this something really dependent on environment, and need user to implement it? Just an advice.

MaJerle, 2021-05-17:
Processing LwESP normally isn't possible from interrupt since mutex cannot be locked from interrupt.

User has to manually setup new thread that calls lwesp_input_process, or if it prefers UART IRQ option, then it can call lwesp_input and data are written to buffer, that is then processed by process thread.

niedong, 2021-05-17:
OK, thanks.

closed 2021-05-17 by niedong