MaJerle/lwesp #93: On the separation of packets in UDP transmission and reception.

TakashiKusachi, 2021-07-22:
In TCP transmission and reception, it is correct to cut packets according to the size of the buffer. However, I don't think it is correct to cut off packets in UDP.
In UDP, I think that an error should be issued in such cases.
It is this code.
https://github.com/MaJerle/lwesp/blob/e4f112f92a4f7618a14b51498be72c65af7aaf39/lwesp/src/lwesp/lwesp_int.c#L476

And the same goes for receiving UDP
It is this code.
https://github.com/MaJerle/lwesp/blob/e4f112f92a4f7618a14b51498be72c65af7aaf39/lwesp/src/lwesp/lwesp_int.c#L1312-L1345

However, the above is true for sockets like POSIX, otherwise there is no need to follow it.

This is machine translation. I'm sorry if the text is difficult to understand.

MaJerle, 2021-12-20:
So you prefer to issue an error if data to send is larger than max available packet?

TakashiKusachi, 2021-12-20:
I think it is desirable to give an error[e.g. EMSGSIZE] by default.