MaJerle/lwesp #108: MQTT over SSL

jondurrant, 2022-01-15:
I required MQTT over SSL so that there is a degree of protection on network. (Not investigated certificate authentication, yet).  

Local fork of code to allow extension of: lwesp_mqtt_client_info_t to include ssl booling flag, and lwesp_mqtt_client_connect to use LWESP_CONN_TYPE_SSL.

I believe you have plans to look at TLS/SSL management, which makes sense. Happy to share my fudge.

MaJerle, 2022-01-15:
Is this a message that you have done it or you are seeking for solution?

For the moment, I have no implementation for secure TCP, but with mbedTLS it is a long term plan

jondurrant, 2022-01-15:
I've made the changes, so code is available (https://github.com/jondurrant/lwespFork)

Security concerns  right now would be in network penetration. If you have got into my home and have the device in your hand I think I have bigger problems;-)

MaJerle, 2022-01-15:
Could you create a pull request of your changes?

jondurrant, 2022-01-15:
Yes of course.

jondurrant, 2022-01-16:
Pull created, though can't work out how to link to issue

MaJerle, 2022-01-16:
I see it. It needs coding style improvementa first, then it can be merged.

Thanks for proposing.

closed 2022-01-16 by MaJerle

jondurrant, 2022-01-16:
I will leave that with you as not sure what coding standards you are
working too.

Jon