MaJerle/lwesp #110: Support for passthrough mode via AT+CIPSEND?

bmkuter, 2022-03-04:
Is there support for entering passthrough mode?
[Documentation here](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#id20)

MaJerle, 2022-03-04:
It is not implemented nor is in the roadmap, because I do believe it has no added value.

Usually when you have a "passthrough mode", as indicated in the docs, it is time-based sending, every 20ms. This is not so much, but if you have plenty of data, it can be challenging.

I have worked in the past on a BLE module through AT commands (SSP protocol) and we used passthrough mode. I can share:

- Throughput was lower than if we would always call "AT+SEND...=len,data,...", even if this has more characters on UART line
- Data handling was more challenging as you never know if "ERROR or EXIT" is coming from your data or because device exited transparent mode.

That being said, I did not plan it, but I would like to hear your opinion, why do you see it would be a must to have it. Then we can reconsider.

bmkuter, 2022-03-04:
Thanks for the response, your reasoning makes a lot of sense. The reason I wanted to use it is to directly pass through whatever data I'm receiving to be processed without pruning the "+IPD" section. For example if I want to print a TCP message from a server to a screen on my MCU. However if the passthrough mode really impacts speed, I can always do this some other way, like strtok().

MaJerle, 2022-03-04:
> Thanks for the response, your reasoning makes a lot of sense. The reason I wanted to use it is to directly pass through whatever data I'm receiving to be processed without pruning the "+IPD" [...]

Or you use this library and output data to your terminal on connection data receive callback. Data are already parsed at that time.

bmkuter, 2022-03-04:
Thank you for the help. Can you elaborate a bit more? I am using the Client RTOS example, but the data output still has the AT command around it. For example, if I send `Hello` from my TCP server, the printed message from the serial connection is `+CIPRECVDATA:6,"192.168.86.123",1000,Hello`. 

I think I may be missing a step if I just want to print the message, `Hello`.

MaJerle, 2022-03-04:
You are right on this aspect, if you only plan to use UART raw output, then you will see `+CIPRECVDATA...`, but library can do post-processing for you.

You can, as you said, do it like you know, but I would suggest you do it with limit in length too, something like:

`printf("pbuf: %.*s\n", (int)lwesp_pbuf_length(pbuf, 0), (char*)lwesp_pbuf_data(pbuf));`

closed 2022-03-10 by bmkuter