MaJerle/lwesp #111: Missing lwesp_sys_template.c?

bmkuter, 2022-03-04:
Reading through the porting guide, I should see a template for filling in my system functions, `lwesp/src/system/lwesp_sys_template.c`? Has that been deprecated? Same question with the template folder missing here: `lwesp/src/include/system/port/template/*`

MaJerle, 2022-03-04:
Seems odd, indeed.

I suggest you take already existing port (win32, or cmsis, or freertos) and adjust it for your needs.

Which arch do you use?

bmkuter, 2022-03-04:
I was planning on using MQX from NXP.

MaJerle, 2022-03-04:
> I was planning on using MQX from NXP.

This one is technically unknown to me. I know it exists but I have no understanding of its API. Sorry, you will have to manually implement it.

bmkuter, 2022-03-08:
Thanks! Your FreeRTOS template should help

closed 2022-03-08 by bmkuter