MaJerle/lwesp #112: Clarification on steps for adding library to project?

bmkuter, 2022-03-10:
I'm trying to add the lwESP library to a project and I was wondering if I could have clarification on the process. Do I need to complete the porting guide steps before adding the library to the project itself? 

I am following the steps from the Getting Started guide, but I am unable to build the project. I am not attempting to actually implement any of the lwESP functions yet in my application, just compile the library alongside my already-working project.

MaJerle, 2022-03-11:
Unless you call `lwesp_init` in your application part, library should compile with no LL port, but you do need to prepare system port, at least header file -> there is a definition of some types used across the lib.

bmkuter, 2022-03-11:
Thanks! So there are 3 main files to configure:
* The LL port 
* lwesp_sys_template.c
* lwesp_sys_port.h

Is there a template file for the LL port, like there is [for ](https://docs.majerle.eu/projects/lwesp/en/latest/user-manual/porting-guide.html#steps-to-follow)the other components?

MaJerle, 2022-03-11:
Correct

closed 2022-03-11 by bmkuter