MaJerle/lwesp #113: Configuring flow control via AT+UART_CUR

bmkuter, 2022-03-11:
I'm looking to enable flow control via this [AT command](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Basic_AT_Commands.html?highlight=baud#at-uart-cur-current-uart-configuration-not-saved-in-flash), and was wondering if there's an existing method? I know there's a function for controlling [baud rate of the AT port ](https://docs.majerle.eu/projects/lwesp/en/latest/api-reference/lwesp/index.html?highlight=baud#_CPPv421lwesp_set_at_baudrate8uint32_tK20lwesp_api_cmd_evt_fnPCvK8uint32_t) in lwESP, but I can't find mention of adjust flow control.

MaJerle, 2022-03-11:
LwMEM currently does the work w/o using flow control.

A new function `lwesp_set_at_baudrate_ex` is expected to be added with flow control capability.

bmkuter, 2022-03-14:
Amazing, thanks!

closed 2022-03-14 by bmkuter