MaJerle/lwesp #125: About support BLE feature of ESP32

xiongyu0523, 2023-04-26:
Hi, Majerle

How do you envision lwesp support of BLE? 

ESP-AT supports both WIFI and BLE on some of the dual mode ESP silicon, if we use lwesp for WIFI, the library takes control of AT commands to send & receive, how can a user use ESP-AT fulfill their BLE needs?

Do you think adding a BLE API to lwesp is feasible or any short-term workaround can be done to add BLE support when using lwesp?

Tizio1234, 2023-10-23:
That would definitely be interesting. How does the lib parser work?
I'd like to try and add something myself

csstraut, 2025-11-12:
Hi @MaJerle, 

With the growth of connected devices, especially around the BLE part, I am really interested about the BLE support for the ESP32 of this library, beside the WiFi usage.

Is there any update related to any planned support for this?

Thank you!