MaJerle/lwesp #126: Memory deallocation problem

HelplessPineapple, 2023-07-06:
First of all, thanks for the great work with LWESP - rarely have I come across such a well-documented and user-friendly open source project.
I am currently using the netconn API and have implemented my code based on the netconn_server snippet.
Although I have been careful to allocate enough memory and free the memory of the pbufs when necessary, I (always) keep running into memory allocation issues after a while. 
Thanks to the library's great logging feature, I can see that the dynamic allocations should be freed correctly. 
 
Here are the end the logs of the pbufs and dynamic memory allocations with the memory allocation failure at the end:
1.
[MSG VAR] Allocated 76 bytes at 20016898
[MSG VAR] Allocated 76 bytes at 200168EC
[MSG VAR] Free memory: 20016898
[MSG VAR] Allocated 76 bytes at 2001685C
[MSG VAR] Free memory: 200168EC
[MSG VAR] Free memory: 2001685C
[MSG VAR] Allocated 76 bytes at 20016874
[MSG VAR] Allocated 76 bytes at 200168C8
[MSG VAR] Free memory: 20016874
[MSG VAR] Allocated 76 bytes at 20016874
[MSG VAR] Error allocating 76 bytes
[MSG VAR] Free memory: 200168C8
[MSG VAR] Free memory: 20016874

2.
[LWESP PBUF] Allocated 49 bytes on 200168EC
[LWESP PBUF] Deallocating 200168EC with len/tot_len: 49/49
[LWESP PBUF] Allocated 49 bytes on 200168EC
[LWESP PBUF] Deallocating 200168EC with len/tot_len: 49/49
[LWESP PBUF] Allocated 49 bytes on 200168B0
[LWESP PBUF] Deallocating 200168B0 with len/tot_len: 49/49
[LWESP PBUF] Allocated 49 bytes on 200168E0
[LWESP PBUF] Deallocating 200168E0 with len/tot_len: 49/49
[LWESP PBUF] Allocated 49 bytes on 20016874
[LWESP PBUF] Deallocating 20016874 with len/tot_len: 49/49
[LWESP PBUF] Allocated 49 bytes on 200168F8
[LWESP PBUF] Deallocating 200168F8 with len/tot_len: 49/49
[LWESP PBUF] Allocated 49 bytes on 20016910
[LWESP PBUF] Deallocating 20016910 with len/tot_len: 49/49
[LWESP PBUF] Failed to allocate 49 bytes
[LWESP PBUF] Failed to allocate 24 bytes

Does anyone have any clues as to what the problem might be? 
I'm grateful for any help! :)

MaJerle, 2023-07-06:
I definitely use netconn API very often and was never able to run into memory allocation issues.

Can you show the code and how much have you allocated for memory?

HelplessPineapple, 2023-07-06:
LWESP_MEM_SIZE is set to 0x4000, so around 16000 bytes. This error **only** occurs if the LWESP-memory is full and not (in my case) the FreeRTOS memory, am I right? 

The netconn-API part is almost identical to the netconn_server-snippet, except that the sending of data is done by another thread. 

That's the processsing thread for the respective TCP-connections: 

```
/**
 * \brief           Thread to process single active connection
 * \param[in]       arg: Thread argument
 */
static void
netconn_server_processing_thread(void* const arg) {
    lwesp_netconn_p client;
    volatile lwesp_pbuf_p pbuf, p = NULL;
    volatile lwespr_t res;
    volatile uint8_t *data_pointer = NULL;

    //printf("Current heap size: %d \n", xPortGetFreeHeapSize());

    client = ((struct tcp_server_input_args*) arg)->client;
    current_client = client;
    if(current_client != NULL){
    	//printf("Overwriting older tcp client!! \n");
    }
    //printf("A new connection accepted!\r\n");   /* Print simple message */
    lwesp_netconn_set_receive_timeout(client, 0);

    do {

    	//osDelay(10);

    	res = lwesp_netconn_receive(client, &pbuf);

        if(res == lwespTIMEOUT){
            if(p != NULL){
    			lwesp_pbuf_free(p);
    			p = NULL;
            }
        	continue;
        }
        if (res == lwespOK) {

            if (p == NULL) {
                p = pbuf;    /* Set as first buffer */

            } else {
            	lwesp_pbuf_cat(p, pbuf);        /* Concatenate buffers together */
            }
            if(p != NULL){
            	tcpMsgHandler_handleReception(p,9,40);
    			lwesp_pbuf_free(p);
    			p = NULL;
            }
        }
    } while (res == lwespOK || res == lwespTIMEOUT);

    current_client = NULL;

    if (res != lwespCLOSED) {
    	lwesp_netconn_close(client);
    }     /* Close netconn connection */
    if (p != NULL) {                            /* Free received data */
        lwesp_pbuf_free(p);
        p = NULL;
    }

    //printf("Connection error: destroy TCP Thread! \n");
    lwesp_netconn_delete(client);                 /* Destroy client memory */
    lwesp_sys_thread_terminate(NULL);             /* Terminate this thread */
}

```
And that's the function called to send TCP data: 

```
static bool tcpMsgHandler_transmitData( lwesp_netconn_p tcp_client,uint8_t* data, uint8_t len){

		volatile lwespr_t res = lwesp_netconn_write(tcp_client,(void*) data,len);
		if(res == lwespOK){
			res = lwesp_netconn_flush(tcp_client);
			return true;
		}
		else{
			//lwesp_mem_free_s((void**)&nc->buff.buff);
			//lwesp_mem_free_s(&pointer);
			printf("Error!\n");
		}
		return false;
}

```
The low-level code is the same as the stm32 template-code.

MaJerle, 2023-07-06:
Someone is allocating memory. If you have 16k, then few allocations of ~70bytes cannot lead to no memory.

HelplessPineapple, 2023-07-06:
It's not a few allocations, that's just the end of the logs. The 76 bytes are allocated and freed every 80ms. And after ~2min the error allocation error occurs. 
For every new TCP reception/transmission and new socket connection is established.

HelplessPineapple, 2023-07-07:
When I log the LWESP Memory, I suppose the threads are so busy with logging that the netconn server does not even get created and initialized. 

Thats the start of the logging output:
Initializing LwESP
Library initialized!
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM[LWESP MEM] Allocatio] Callocatn OK: 76 bion OK: 16 ytes, addrbytes, addr: 20012990: 200129E4

[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
[LWESP MEM] Callocation OK: 16 bytes, addr: 20012978
[LWESP MEM] Free size: 16, address: 200129E4
[LWESP MEM] Callocation OK: 16 bytes, addr: 200129E4
[LWESP MEM] Free size: 16, address: 20012978
.
. 
.

Is it normal that the lib communicates with the ESP so often during initialization?

MaJerle, 2023-07-07:
Yes, there are many commands that are sent out to device during the startup. `16` bytes allocation seems to be from `lwesp_timeout_add` function call, where structure is `16` bytes on `32-bit` systems.

HelplessPineapple, 2023-07-07:
Yes, you are right. Because only the _putchar method itself is thread-safe, so if two threads want to write at the same time, that happens. 
But other than for readability reasons, is there a problem with that?

MaJerle, 2023-07-10:
In recent days, due to wrongly configured low-level communication, we saw some cases where memory was not reset properly. Netconn API has been updated, adding connection val_id in the netconn structure.

HelplessPineapple, 2023-07-11:
Thanks for your help ! :)
Do you know which part of the low level communication part could be misconfigured? 
I used exactly the same configuration as in the lwesp_ll_stm32.c-file.

HelplessPineapple, 2023-07-18:
Can you give me a hint which part of the low-level communication is misconfigured?
With the latest versions (with "lwesp_pub_s()") the memory allocation error still occurs - always at the same time and with the same number of bytes with my test program. So at least the behavior of the bug is deterministic.

HelplessPineapple, 2023-07-18:
Sure!

With the exception of MEM (which stalls the program flow because of its massive amount of debug MSGs), I have logged extensively with PBUF and MSG, but could not find any hints  of any memory leaks.

MaJerle, 2023-07-27:
>I don't really understand why sometimes 80 bytes are freed when there were never exactly 80 bytes allocated.

This is because I am printing the user input data size len (`77` bytes`) while when free is called, debug prints actual size of allocation, which was `80` bytes. Reason is that library uses memory alignment, by default set to `4` bytes.

It would be better to update the library to print actual allocation to not confuse you.

What I see clearly is that `2048` blocks are allocated and freed afterwards and you are still entering the memory failure?
Do you have some custom `lwesp_timeout` feature?

We will need a computer debugging. I'm not able to see an issue from this.

HelplessPineapple, 2023-07-27:
No, I only use the standard lwesp functions. 
Sure, that would be great :) 
Should I contact you by email regarding the debugging session?

closed 2023-10-01 by MaJerle