MaJerle/lwesp #129: Not getting any output on stm32 uart port

Tizio1234, 2023-09-05:
I'm using an stm32f411re nucleo board, I added and ported the library to my project following documentation, this is the stm32 board specific setup I'm using:
`/* USART */
#define LWESP_USART                           USART6
#define LWESP_USART_CLK                       LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_USART6)
#define LWESP_USART_IRQ                       USART6_IRQn
#define LWESP_USART_IRQHANDLER                USART6_IRQHandler
#define LWESP_USART_RDR_NAME                  DR

/* DMA settings */
#define LWESP_USART_DMA                       DMA2
#define LWESP_USART_DMA_CLK                   LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_DMA2)
#define LWESP_USART_DMA_RX_STREAM             LL_DMA_STREAM_1
#define LWESP_USART_DMA_RX_CH                 LL_DMA_CHANNEL_4
#define LWESP_USART_DMA_RX_IRQ                DMA2_Stream1_IRQn
#define LWESP_USART_DMA_RX_IRQHANDLER         DMA2_Stream1_IRQHandler

/* DMA flags management */
#define LWESP_USART_DMA_RX_IS_TC              LL_DMA_IsActiveFlag_TC1(LWESP_USART_DMA)
#define LWESP_USART_DMA_RX_IS_HT              LL_DMA_IsActiveFlag_HT1(LWESP_USART_DMA)
#define LWESP_USART_DMA_RX_CLEAR_TC           LL_DMA_ClearFlag_TC1(LWESP_USART_DMA)
#define LWESP_USART_DMA_RX_CLEAR_HT           LL_DMA_ClearFlag_HT1(LWESP_USART_DMA)

/* USART TX PIN */
#define LWESP_USART_TX_PORT_CLK               LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_GPIOC)
#define LWESP_USART_TX_PORT                   GPIOC
#define LWESP_USART_TX_PIN                    LL_GPIO_PIN_6
#define LWESP_USART_TX_PIN_AF                 LL_GPIO_AF_8

/* USART RX PIN */
#define LWESP_USART_RX_PORT_CLK               LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_GPIOC)
#define LWESP_USART_RX_PORT                   GPIOC
#define LWESP_USART_RX_PIN                    LL_GPIO_PIN_7
#define LWESP_USART_RX_PIN_AF                 LL_GPIO_AF_8`
Is there something wrong with this setup? Also, this is my init_thread:
`/* USER CODE BEGIN Startinit_thread */

	printf("Initializing LwESP\r\n");
		while(1){
			if (lwesp_init(NULL, 1) != lwespOK){
				printf("Failed initializing LwESP, retrying\r\n");
				osDelay(1000 / portTICK_PERIOD_MS);
			} else {
				printf("Correctly initialized LwESP\r\n");
				break;
			}
		}

		printf("Joining AP\r\n");
		while(1){
			if (lwesp_sta_join(ssid, password, NULL, NULL, NULL, 1) != lwespOK) {
				printf("Failed joining %s, retrying\r\n", ssid);
			} else {
				printf("Joined %s\r\n", ssid);
			}
		}

		lwesp_ip_t sta_ip;
		lwesp_sta_getip(&sta_ip, NULL, NULL, NULL, NULL, 1);

		printf("Station IP: %i:%i:%i:%i", sta_ip.ip[0], sta_ip.ip[1], sta_ip.ip[2], sta_ip.ip[3]);

  /* Infinite loop */
  for(;;)
  {
    osDelay(1);
  }
  /* USER CODE END Startinit_thread */`
  it never gets out of the first while loop, because I don't get any output on the stm32 usart6.
  Any idea on what I did wrong?

MaJerle, 2023-09-05:
Does STM32 send any data out at all? Please check your low-level implementation, which is the responsibility of an implementaiton

Tizio1234, 2023-09-06:
> Does STM32 send any data out at all? Please check your low-level implementation, which is the responsibility of an implementaiton

I'm using lwesp_ll_stm32.c file and including the board specific file(f411re) I included(sorry for bad format), thank you for your immediate response.

Tizio1234, 2023-09-06:
I noticed now I had attached my usb to serial converter to the wrong pins, sorry for wasting your time, now that I have connected it correctly I do get an output.

closed 2023-09-06 by Tizio1234