MaJerle/lwesp #132: Problem with initializing LwESP library on stm32f746g discovery kit

Tizio1234, 2023-09-17:
I'm trying to setup LwESP on the st3m2f746g discovery kit, the whole setup process went fine for the stm32f411re nucleo board, but here I'm having trouble initializing the library.
I'm calling lwesp_init from a thread I created(stack size is 256 in words, for the priority I've tried Low, BelowNormal, Normal and AboveNormal but the result was always the same), after approx. 10 seconds it fails with 5 as the return value.
I modified the lwesp_input.c file to see what the st3m2 was receiving on the uart, and to me, it looks fine(the data inside parenthesis after "Received data: " is the one coming from the esp uart, the rest is just messages for debugging, I'm using the example common global callback function):
Event no. 0
Library initialized!
Received data(AT+RST

OK
)
Event no. 14
Received data(
ready
)
Event no. 1
Device reset detected!
Received data()
Received data()
Event no. 14
Received data()
Event no. 14
Event no. 14
Event no. 14
Event no. 4
Event no. 3
Received data(AT+RST

OK
)
Event no. 14
Event no. 14
Received data(
ready
)
Event no. 1
Device reset detected!
Received data()
Received data(AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g24b9d38a2)
Received data(4-dirty
compile time(6118fc22):Jul 28 2023 09:47:28
Bin version:v3.2.0.0(WROOM-32)

OK
)
Received data(
OK
)
Received data()
Event no. 14
Event no. 14
Event no. 14
Event no. 4
Event no. 2
Failed to initialize LwESP, error code: 5

I'm using an ESP32 and I added a printf to the example global function that prints the event number(before the possible event log), am I doing something wrong?

MaJerle, 2023-09-17:
Low level between F411 and F746 is more or less the same, with one big excetpion. Cortex-M7 has cache, so if enabled, and DMA is used, you may have transparency issues.

Tizio1234, 2023-09-17:
So what should I disable/enable or modify in order to get the library working?

Tizio1234, 2023-09-22:
Sorry for being very generic, now I've done more tests:
In cubemx, section "CORTEX_M7", I can choose to enable or disable instruction and data caches:
if I keep them both enabled, touchgfx works, but the library does not seem to be responding, in the sense that lwesp_input_process function does get called correctly(as I've shown in the first "comment") but I get lwespTIMEOUT on lwesp_init;
if I disable instruction cache, the end result is the same as keeping them both active;
if I disable data cache, and disable or enable(doesn't change the end result) instruction cache, after programming the board, it looks like everything is working, except TCP connections, but I'm not sure, I need to do further testing; instead, if I turn off and on the stm32 board, from then on the board gets in HardFault, it never shows the touchgfx screen, just a repeating pattern in the display.
But, the esp32 does get reset, so it means the init_thread had started, I have no idea on what the problem could be, any help? Thanks in advance to the people that are willing to help in any way

MaJerle, 2023-09-22:
From your description, it seems you are indeed a victim of a cache and DMA coherency.
You may need to study:

- How cache works in Cortex-M7 (WT, reload, ...)
- How to configure MPU to control cache.

I propose you to, then:

- Study how to place memory buffer, used for RX DMA buffer (lwesp DMA buffer) in its own section
- Put this section to specific memory address
- Disable the cache for that area

STM32CubeF7 or STM32CubeH7 ethernet examples could help you with that, since there some of the DMA buffers are placed in the proper areas and memory protection unit is to define cache policy for these regions.

You may also read this: https://www.youtube.com/watch?v=6IUfxSAFhlw&ab_channel=STMicroelectronics

Tizio1234, 2023-09-23:
Thank you for your time and answer, should I close the issue while studying what you mentioned and trying to solve the problem?

Tizio1234, 2023-09-23:
> From your description, it seems you are indeed a victim of a cache and DMA coherency.
> You may need to study:
>
> - How cache works in Cortex-M7 (WT, reload, ...)
> - How to configure MPU to control [...]

So the library cannot access the usart_mem buffer, but the uart thread can?

MaJerle, 2023-09-23:
From your prints, it doesn't seem that you received all data. Can you print data when cache is enabled and when it is not, based on your 3 cases?

Tizio1234, 2023-09-23:
> From your prints, it doesn't seem that you received all data. Can you print data when cache is enabled and when it is not, based on your 3 cases?

What do you mean by not all data?
When the Dcache is enabled, I receive the data in the prints between two parentheses after the keyword "Received";
when it's not enabled, the stm32 board crashes because of touchgfx not being able to work without Dcache, from what I know, this happens since version 4.16, I'm using version 4.20.

MaJerle, 2023-09-23:
Touchgfx doesnt care about cache. You can disable it at the moment while we debug the esp.

So please, try cache enabled or disabled and compare the print in the Received part. Are you getting the same? Id like to see log of all 3 cases you said before.

Tizio1234, 2023-09-23:
I don't know if it's my specific board or something else, but if I disable D cache, touchgfx does not work and the board goes into hardfault, I tried making a new touchgfx project with a slider and then disabling the D cache with cubemx, when they were enabled, it was working, but then once I uploaded the program with cache disabled and turned off and on the board, it crashed into hardfault, am I doing something wrong?

MaJerle, 2023-09-23:
@Tizio1234 I am not able to help you if you do not listen. Please disable in the code the part, that initialized TouchGFX. If you use TOuchGFX designer, you certianly have MX_TouchGFX_Init() in the main file.

Tizio1234, 2023-09-23:
I'm sorry for not listening, here are the logs I got, with the cache disabled, and then, enabled, should I also log data sent from the stm32?
cache disabled:
Device reset detected!
Received(AT+RESTORE
)
Received(
OK
)
Event no. 14
Event no. 14
Received(
ready
)
Device reset detected!
Received(ATE0

OK
)
Received(AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g2)
Received(4b9d38a24-dirty
compile time(6118fc22):Jul 28 2023 09:47:28
Bin version:v3.2.0.0(WROOM-32)

OK
)
Received(
OK
)
Received(
OK
)
Received(
OK
)
Received(
OK
)
Received(+CWDHCP:3
OK
)
Received(
OK
)
Received(
OK
)
Received(
OK
)
Received(STATUS:1

OK
)
Received(
OK
)
Received(+CIPAP:ip:"192.168.4.1"
+CIPAP:gateway:"192.168.4.1"
+CIPAP:netmask:"255.255.255.0"

OK
)
Received(+CIPAPMAC:"94:3c:c6:37:ed:e5"

OK
)
Received(
OK
)
Event no. 3
Received(
OK
)
Event no. 14
Event no. 14
Received(
ready
)
Device reset detected!
Received(ATE0

OK
)
Received(AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g2)
Received(4b9d38a24-dirty
compile time(6118fc22):Jul 28 2023 09:47:28
Bin version:v3.2.0.0(WROOM-32)

OK
)
Received(
OK
)
Received(
OK
)
Received(
OK
)
Received(
OK
)
Received(+CWDHCP:3
OK
)
Received(
OK
)
Received(
OK
)
Received(
OK
)
Received(STATUS:0

OK
)
Received(
OK
)
Received(+CIPAP:ip:"192.168.4.1"
+CIPAP:gateway:"192.168.4.1"
+CIPAP:netmask:"255.255.255.0"

OK
)
Received(+CIPAPMAC:"94:3c:c6:37:ed:e5"

OK
)
Received(
OK
)
Event no. 2
Initialized LwESP correctly
Event no. 14
Event no. 14
Event no. 14
Event no. 14
Event no. 14
Event no. 14
Event no. 14
Event no. 14
Event no. 14
Event no. 14

cache enabled:
Initialized LwESP library component
Received(
ready
)
Device reset detected!
Received(AT+RESTORE
)
Received(
OK
)
Event no. 14
Event no. 14
Received(
ready
)
Device reset detected!
Received(ATE0

OK
)
Received()
Received(b9d38a24-dirty
compile time(6118fc22):Jul 28 2023 09:47:28
Bin version:v3.2.0.0(WROOM-32)

OK
)
Received()
Received(
OK
)
Received()
Received()
Event no. 14
Event no. 14
Event no. 14
Event no. 4
Event no. 3
Received(
OK
)
Event no. 14
Event no. 14
Received()
Event no. 14
Event no. 14
Event no. 14
Event no. 4
Event no. 2
Failed to initialize LwESP, error code: 5

MaJerle, 2023-09-23:
As you can clearly see, when cache is enabled, you are not really receiving all the data. This indicates that your memory protection unit isn't properly set and therefore cache invalidation is not handled properly.

3 options you can do:
- Disable cache globally, but then check where does touchgfx low-level driver call SCB_InvalidatexCache or SCB_CleancCache (where `x` can be `I` or `D`) and delete these calls.
- Disable cache just for your dma usart buffer. This involves putting the usart_mem buffer into memory address (through linker sections might be the easiest way) and configure Memory Protection Unit to not apply cache there
- Call SCB_InvalidateDCache(usart_mem, sizeof(usart_mem)) before calling `lwesp_input_process`. This one may work.

Sorry, but more than this, it is beyond this library.

Tizio1234, 2023-09-24:
Thank you for your help, I've tried the third options, and now it's working, what are the pros and cons of the three options you mentioned?

MaJerle, 2023-09-24:
Take your time and read this: https://www.st.com/resource/en/application_note/an4839-level-1-cache-on-stm32f7-series-and-stm32h7-series-stmicroelectronics.pdf

closed 2023-09-24 by MaJerle