MaJerle/lwesp #134: Feature request: ability to start connections/transmissions when esp is in AP mode

Tizio1234, 2023-09-29:
At the moment, lwesp_conn_start(...) does not work when the esp is in only AP mode, it can only act as a server, this completely removes the ability to use UDP. If lwesp_conn_start(...) gets called with LWESP_CFG_STATION_MODE defined to 0, it returns error 2, if LWESP_CFG_STATION_MODE is defined to 1,  returns error 9(station does not have IP). Because of this, also netconn API does not work, this makes applications in AP mode harder to write and UDP protocol impossible to use.
Even though, esp-at does support all features when in AP mode.

MaJerle, 2023-09-29:
2 commits:
- do not return just error, rather return cmd not supported, if it cannot be started due to conditional kick-ou https://github.com/MaJerle/lwesp/commit/99c1ac3a70a203cfebdaad29a335c933f5cf2f29
- Add option to start connection w/o checking IP status. You have to enable the feature in `lwesp_opts.h` configuration file: https://github.com/MaJerle/lwesp/commit/99c1ac3a70a203cfebdaad29a335c933f5cf2f29

Simply pull latest change and try. let me know if that works.

Tizio1234, 2023-09-30:
I was reimporting the library, this time copying with git so I can update it easily, what about line 222 of lwesp_ll_stm32.c that broke stm32f4 and stm32f7 driver?

MaJerle, 2023-09-30:
Let's use this issue # for new feature, and we discuss F7/H7 driver in separate thread. I agree there is a need to update generic driver.

Tizio1234, 2023-09-30:
Regarding UDP connection, the connection does start correctly, but I'm still getting lwespERRPAR on lwesp_conn_send(...):
Device reset detected!
Device reset detected!
Initialized LwESP(and ESP) correctly
Set ap only mode
Global Event no. 14
Configured AP correctly
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
Global Event no. 14
New station connected to access point with MAC address: E8:84:A5:8E:60:8E

IP 192.168.4.2
 assigned to station with MAC address: E8:84:A5:8E:60:8E

Connection 0 active!
Started UDP connection correctly
Global Event no. 14
Connection Event no. 12
[LWESP CONN] Poll event: 0x200056a4
Connection Event no. 12
[LWESP CONN] Poll event: 0x200056a4
lwesp send result: 3
Global Event no. 14
Connection Event no. 12
[LWESP CONN] Poll event: 0x200056a4
Connection Event no. 12
[LWESP CONN] Poll event: 0x200056a4
lwesp send result: 3
Global Event no. 14
Connection Event no. 12
[LWESP CONN] Poll event: 0x200056a4
Connection Event no. 12
[LWESP CONN] Poll event: 0x200056a4
lwesp send result: 3
Global Event no. 14

The main.c is the same I had sent to you via email, you had mentioned that I wasn't using udp_conn correctly, what am I doing wrong?

MaJerle, 2023-09-30:
Why don't you open debug and analyze why you have *parameter* error, as suggested by email? :)

You did not assign any value to `udp_conn` variable.

closed 2023-09-30 by MaJerle