MaJerle/lwesp #138: Issue in documentation of SSL connection

VinFar, 2023-11-18:
I noticed a missing hint in the documentation of setting up SSL connection with CA file (and probaly also for private key and cert file). This is more an issue of the ESP documentation for generating PKI Files using espressif AtPKI.py python tool for converting .crt files into binary files.

When using the AtPKI python tool it, additionally to converting it to a binary, adds some information to the binary file. If one uses the generated binary directly with the function `lwesp_mfg_write` to write it to the NVS, the stored certificates does not work. Rather, you have to strip the added header and footer of the binary file.

The generated file looks like this:
`0xF1, 0xF1, 0x01, 0x00, 0x98, 0x07, 0x00, 0x00, 0x01, 0x00, 0x93, 0x07, 0x2D, 0x2D, 0x2D, 0x2D, 
0x2D, 0x42, 0x45, 0x47, 0x49, 0x4E, 0x20, 0x43, 0x45, 0x52, 0x54, 0x49, 0x46, 0x49, 0x43 .... more bytes following `.
The issue is, that the bytes until the first `0x2D` are wrong and needs to be removed, leading to:
`0x2D, 0x2D, 0x2D, 0x2D, 
0x2D, 0x42, 0x45, 0x47, 0x49, 0x4E, 0x20, 0x43, 0x45, 0x52, 0x54, 0x49, 0x46, 0x49, 0x43 ....`
The same with the last byte `0xFF`, that also needs to be removed.

If these bytes are removed, the certificate works as expected.

Maybe you can add this information to your documentation, as I wasted some time figuring this out.

Nevertheless, your work is amazing! I am using your library for a IoT product for our start-up! Keep up the great work!

VinFar, 2023-11-18:
Nevermind, the certificate does not need to be conerted using the AtPKI tool, as the certification needs to loaded in its original ASCII format to the NVS :)

MaJerle, 2023-11-18:
> Nevermind, the certificate does not need to be conerted using the AtPKI tool, as the certification needs to loaded in its original ASCII format to the NVS :)

Exactly, with the latest MFG updates, it is directly copy-paste. In the past, with AT+SYSFLASH, it was necessary to use ATPKI and then it has prefix in generated data.

closed 2023-11-18 by MaJerle