MaJerle/lwesp #141: Trouble with establishing TCP/SSL connection

fahirad, 2025-04-01:
Hello. 
So I see that when I call a function to establish the connection, the subcommand is cipstate. It feels troublesome. Because, this is what happens.
CIPSTATE is sent, return is OK, CIPSTARTEX is sent, return is ok, CIPSTATE is sent, +CIPSTATE message and OK is returned and yet I still get a failed connection. 
Not sure if that really is the problem, but is it possible that esp.m.conns and esp.m.active_conns are running around for each other and that causes the issue? 
Has anyone had this and can anyone help? 
Thanks a lot!

MaJerle, 2025-04-01:
To really connect with "SSL" connection, you need to setup the SSL parameters before.

Can you share the API calls and the log AT command info?

fahirad, 2025-04-02:
Not sure if SSL parameters are the issue because I am trying to establish a tcp connection and same thing happens, it seems that it connects but it is still considered disconnected by LwESP
.
I am using the netconn tcp client example.  I don't have access currently to detailed esp-at log, but I can sniff the commands and returns from them, if this is helpful. 
Also wondering why does initiating a connection require sending CIPSTATE command before and after? Btw, this is just a server I created using a phone app, everything is connected to the same WiFi, I tried connecting to the server via a simple python tcp client and that worked fine. And thanks for the quick response. 
                                                           
 -------------AT+CIPSTATE?

                                                           
 -------------OK

                                                           
 -------------AT+CIPSTARTEX="TCP","10.186.53.23",8080,0

                                                           
 -------------CONNECT
                                                         

                                                           
 -------------OK
                                                              
                                                           
 -------------AT+CIPSTATE?

                                                           
 -------------+CIPSTATE:0,"TCP","10.186.53.23",8080,55986,0
                   
                                                             

                                                           
 -------------OK
                                                    
 -------------Cannot connect

fahirad, 2025-04-02:
Also, I have to say that in the parser for CIPSTATE message this if statement evaluates to true
```
if (!esp.m.conns[cn_num].status.f.active) {
        lwesp_conn_close(&esp.m.conns[cn_num], 0);
    }
```

This lead me to think that those status flags might have something to do with the problem.

MaJerle, 2025-04-02:
The parsing if connection is alive or not is done via `+LINK_CONN` message from the module. We'd need to analyze this.
And what versions of AT you have?

fahirad, 2025-04-03:
I am using 3.2.0.0 version and ESP32-C3

fahirad, 2025-04-03:
I see that the reset sub commands were not all sent as they should have been in our project, SYSMSG is one of them, and I see that it configures the +LINK_CONN message. I will try to fix this now. 

Thank you a lot for your help.

closed 2025-05-27 by fahirad